Ultimaker/Cura #14925: Profiles disappears when changing materials

adetogni, 2023-03-17:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Ender S1

### Reproduction steps

When modifying materials, all the printing profiles disappears and you have to recover from backup. To reproduce
- open Manage Materials and create a new one
- for some strange reason appears the Change/Keep changes. Choose Discard
- start creating a new material of an existing brand
- as soon as you move around in the different fields, your profiles disappears and you see just "not supported"
- the only way to restore is to load from a backup

here's a video

### Actual results

profiles disappear

### Expected results

shouldn't happen

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Here's a video

https://icecreamapps.com/v/e3cvakb

[cura.zip](https://github.com/Ultimaker/Cura/files/11000729/cura.zip)

MariMakes, 2023-03-17:
Hey @adetogni,

Welcome to the Ultimaker Cura Github 🚀 

Do I understand correct that you want to save the settings that you changed here?
The ones with the blue reset arrow in front of it? 
[image]

Those settings are usually not saved in a material. 
Those settings are saved in a profile. If you click the save icon you can create a profile from your changed settings. 
[image]

That's how a lot of other people do it. 
I usually save a project file of the settings that I liked for a certain purpose. 
I will load the project file again, remove the model and add the model I want to work with that day. 
To save a project file go to File -> Save project. 

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

adetogni, 2023-03-17:
look at min 1:42 you'll see "not supported" in profiles, whereas a second before you can see "Standard New"

MariMakes, 2023-03-20:
Hey @adetogni,

I apologize I must have misunderstood your report. 

This has been a known issue for a while most interactions around it can be found here https://github.com/Ultimaker/Cura/issues/8229

You can read more about it in the linked ticket but it's actually intended.
The Material Type Field needs to match EXACTLY the known materials in Cura.
In your case, it should be TPU95A instead of TPU. 

But the behavior has been so confusing that we have a ticket on our backlog with the intent to improve this.
We've been getting a lot of reports on this, so I hope we can resolve this soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/8229

closed 2023-03-20 by MariMakes

adetogni, 2023-03-20:
thanks for the feedback, That ticket has A LOT of interactions, so yes, it's super misleading since it's not mentioned anywhere that it should be TPU95A and you're also getting an error. I'll track that one