Ultimaker/Cura #14935: Missing skin/top layer behind top feature or text

cutmoney, 2023-03-17:
### Application Version

5.3.0

### Platform

Mac

### Printer

Ender 3 Max Neo

### Reproduction steps

1. Create a model with a feature which extends above the top of the model (e.g. text on a flat background).
2. Set "Top Surface Skin Layers" to a value greater than zero (e.g. 1)
3. Modify any of the parameters of the skin layer, such as line width or line directions.

### Actual results

Cura masks out the above feature with an outline in the top skin If I change the line width to anything other than the default .40 or the line direction to [90], then Cura masks out the above feature with an outline in the top skin, which creates a much rougher face as it's basically stopping and starting around all of the letters. I've tried every setting imaginable (Skin Overlap Percentage, Expand Distance, Removal Width, etc.) which seems to have zero effect. The picture is with a .28 line width for the skin, the bottom is with a .40 line width.

### Expected results

It should retain a flat surface without any masking below the text.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

cutmoney, 2023-04-07:
I've determined what the root cause of the issue is. If the top feature is large enough that it would require infill, that infill will carry all the way up beyond the skin. So, for text (like pictured above) it will leave weird ghosting in the skin to carry the infill up to the top. I can upload a sample project if you want, but it is consistent on everything I've tried.

Obviously with a large feature, such as a step or wall, this is the desired outcome. But with smaller features, such as text, patterns, or other extruded graphics on a face, it leaves numerous holes and gaps on the top layers (skin) which severely increases the direction changes, time, and lowers the quality substantially. I've attached another couple of pictures as an example. The weird part is changing the line width and layer height does seem to affect it in different ways. Sometimes it happens on a few of the top layers, other times none. Ideally, you could (whether by preference or otherwise) ignore the layers below on the top layers/skin.

[image]
[image]

MariMakes, 2023-05-03:
Hey @cutmoney,

Thanks for the update.

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15028
We have a ticket on our backlog with the intent to improve this behavior.
We see improved behavior if you increase the Skin Expand Distance. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15028
Thanks for your report 👍