Ultimaker/Cura #14938: Tree Support Slicing fails with model

tbnorris, 2023-03-17:
### Application Version

5.3.0

### Platform

Pop!_OS 22.04 LTS

### Printer

Ender 3 V2

### Reproduction steps

1. Setup: Download https://www.printables.com/model/152248-ender-3-v2-squash-ball-feet normal 0.2 PLA settings, select Tree supports 
2. Print any of the v2_ender3v2_*_raised.stl feet besides v2_ender3v2_left_pcu_foot_raised.stl with tree supports Slice success
3.  Remove all elements and place v2_ender3v2_left_pcu_foot_raised.stl and tree supports Slice.  This reports an error.

### Actual results

Fails to slice when Tree support selected.

### Expected results

Successfully slices model with tree supports.  This model is very much a mirror of right model and it works properly.
left model does slice when normal supports are selected.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/11006543/cura.log)
[badslicingleftfoot.zip](https://github.com/Ultimaker/Cura/files/11006548/badslicingleftfoot.zip)

MariMakes, 2023-04-03:
Hey @tbnorris,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

The new Cura 5.3 contains the fix for this for a number of slicing crashes. 
I was able to slice your model successfully in 5.3 Stable.
[image]

Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

If you still run into a crash, please double-check if you shared the correct file and let us know.
I'll take another look at what is going wrong. 👍

tbnorris, 2023-04-03:
I can verify that 5.3.0 on Windows 11 seems to be working but not on the reported platform the non modern Linux AppImage of 5.3.0 does not work.  I downloaded it anew and re-integrated it and it still created the same error.

MariMakes, 2023-04-05:
Hey @tbnorris,

I recognize this line in your log: Engine process is killed. Received return code -11
That seems to be Linux Specific. I'll have one of our Linux Developers take a look.
Fingers crossed 🤞

nallath, 2023-07-25:
I've just tried this with 5.4 on linux and it sliced. Could you try again?

closed 2023-08-08 by github-actions[bot]