Ultimaker/Cura #14942: SLicing causes PC to Blue screen

Gtrshop, 2023-03-18:
### Application Version

5.3.0.

### Platform

W10

### Printer

Ender 5

### Reproduction steps

Added 4 files for a printer cooler to a project
- Sliced
- Crash

Redid this, but SAVED as a project BEFORE slicing
- Sliced
- Crashed

### Actual results

CRASHED out to Bluescreen

### Expected results

Files should be silced so I can save and print
[MINIMUS- DUAL - my Parts.zip](https://github.com/Ultimaker/Cura/files/11007313/MINIMUS-.DUAL.-.my.Parts.zip)

[stderr.log](https://github.com/Ultimaker/Cura/files/11007298/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/11007299/stdout.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I'm new to Cura, hope its not something Im doing incorrectly...

uski, 2023-03-19:
Your PC should never blue screen due to an activity in an application, especially on Windows 10 (unlike older Windows versions like 95 where the kernel was much less protected).

You are most likely experiencing either:
- A memory issue
- A kernel driver bug, most likely the display driver
- Or less likely, a CPU issue

Please try the following and report back:
- Update your display driver. If no update is available and you are using a discrete GPU, try with a different GPU, just to see if it helps.
-  Try running memtest86 or a similar memory testing program (it is possible your computer has one available in the BIOS).

Gtrshop, 2023-03-19:
Well I've not had an issue in the past.. This was my gaming rig and it was rock solid. 

I did investigate a couple of the fail codes, and based on that I did a full uninstall (DDU) of my Nvidia drivers. Ran the windows mem test and it passed, I am planning g on running something a bit more through this evening. Again with ram, I pulled one stick and then the other, I didn't really notice one stick was more problematic than the other. It's decent ran Kingston Fury. 

I have sliced a few other things today, with no issue that I can recall. But that project I sent in... Blows up every time. . 

⁣Get BlueMail for Android ​

uski, 2023-03-19:
I'll let other chime in - I am not saying there is no bug in Cura, just that a bug in an application should never make the computer BSoD, so if there is a bug in Cura, there is also a problem somewhere else, either in a driver or hardware.

Hopefully someone finds something. Do you have more details about the crash? Maybe in the even viewer of Windows? Would be useful to know the exact type of crash and any stacktrace or other information available.

Gtrshop, 2023-03-19:
> crash? Maybe in the Event Viewer of Windows? Would be useful to know

No, I haven't opened that file again as I've had other things to do. I did poke around in event viewer though, couldn't really. Pin it down to anything... "unexpected shutdown" but nothing remarkable anywhere immediately before that. I will keep vigilant.... 😉

⁣Get BlueMail for Android ​

closed 2023-04-21 by github-actions[bot]