Ultimaker/Cura #14943: Multiple seams/defects in outer walls, unique to 5.3

cutmoney, 2023-03-18:
### Application Version

5.3.0

### Platform

Mac

### Printer

Ender Neo Max

### Reproduction steps

1. Moved from Cura 5.2.2 to 5.3.
2. No changes to settings or file whatsoever

### Actual results

I'm now seeing a lot of defects which look like seems on my outer walls on a model that I've printed many of. As shown in the photo, the bottom two prints are from 5.3, the other top ones were printed yesterday on 5.2.2. No changes to settings or the model.

### Expected results

A single outer wall seam, now there are probably about 6-12 of them.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

darrellenns, 2023-03-26:
I'm seeing the exact same thing. This is a regression in 5.3.0 - the problem does not appear in 5.2.2. It seems to happen when a wall goes from straight to large radius convex curve. There is a weird extra move in the gcode output. It's very hard to see in Cura's preview, but is quite clear if the gcode is viewed in the PrusaSlicer viewer.

Here is an example:
[image]

In Cura's preview, you can see a slight discontinuity:
[image]

The resulting print:
[image]

The same STL sliced with Cura 5.2.2 does not have these artifacts. Prusa Slicer also does not have this issue. Note that this is not the Z seam - that's in a different location. This artifact appears in 4 places on this model - every location where a straight wall connects to a large radius convex curve. It does not appear on the small radius curves (small fillets) or on the concave curves.

Here are is the STL and gcode I used:
[STL File](https://mega.nz/file/tc5VUZIS#LIH4um4DLUlyXwsaDhgTzVZCoBlZIm4K8e-69ub1Rik)
[5.3.0 gcode](https://mega.nz/file/tABAjArS#Ikspy-CrnoSEGB7TuZhF4yJPt4lbmFpPwtqq9lb88eY)
[5.2.2 gcode](https://mega.nz/file/gYhw0DIR#1FExquIXyAwgQOXUMzQsennbgpMRIM2JurKCpg6_Mg8)

Both gcode samples were sliced with the Cura default standard 0.2mm profile.

MariMakes, 2023-05-05:
Hey @cutmoney,

Thanks for your report 👍
That doesn't look good 😞 

I agree with @darrellenns, this seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14811
We have one of our developers looking at that as we speak, it seems to be related to microsegments that are not being filtered out correctly. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14811

closed 2023-05-05 by MariMakes