Ultimaker/Cura #14945: Dual extrusion Skips layers per print core

GoguyT3D, 2023-03-18:
### Application Version

5.3

### Platform

Windows 11

### Printer

Ultimaker S3

### Reproduction steps

A dual extrusion print skips every other layer for the support core

### Actual results

Failed print

### Expected results

Both nozzles print on each layer in order

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Dual Carling Switch Mount.zip](https://github.com/Ultimaker/Cura/files/11008474/Dual.Carling.Switch.Mount.zip)

GoguyT3D, 2023-03-18:
[quality_changes.zip](https://github.com/Ultimaker/Cura/files/11008527/quality_changes.zip)
heres the log files

GoguyT3D, 2023-03-18:
The "eraser" next to the model was supposed to "induce" the support nozzle to start the prime tower at the beginning of the print though the support extruder skips every layer. Remove the "eraser" and you will see that the support core does not start until 5.1mm when the cura generated supports start.

GoguyT3D, 2023-03-18:
Don't know if this will help for those of you pythons out there trying to code this fix but removing the prime tower and all unnecessary "erasers" induced the print cores to both print for each required layer

MariMakes, 2023-04-03:
Hey @GoguyT3D,

Good to see you again. 😊 
Sorry, it took us a while to get back to you 😞 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15078
We have resolved the issue, and you can even download a nightly to see for yourself. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15078

closed 2023-04-03 by MariMakes