Ultimaker/Cura #14948: Unnecessary travels

orbitalr, 2023-03-18:
### Application Version

5.3

### Platform

Windows 11 22H2

### Printer

Custom

### Reproduction steps

Non-uniform cylinder shelled to 1.8mm wall thickness, theres some additional feature details bit i dont think thats important as yet. 
0.6mm nozzle, material TPU. 
Profile settings: no retraction, combing not on outer surface, 2 walls, no infill, print think walls: no, optimize wall printing order: no, infill travel optimisation: no, layer hight: 0.3mm, wall ordering: inside to outside.  
Sliced. 
Reviewed results. 

### Actual results

Getting a lot of travel moves that dont appear to result in a print action. Its literally a move from the finish at the end of the layer, then at the start of the next layer a move back to the start point (which was the last layer finish point). 

### Expected results

The head should move from the finish of the last layer to the start of the next layer without visiting another location on the way. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[UnnecessaryTravel.zip](https://github.com/Ultimaker/Cura/files/11009642/UnnecessaryTravel.zip)

Ok, in some examples in my file the iner wall has started in a different location to the outer wall start/finish. There is a ligitimate move here to get from the inner wall finish to the outer wall start. 

Take for example between layers 105 and 106. 
105: Start and finish for all print moves are in the same place. 
 At the end of the layer the head moves to a different location. 
106: The head is where this ^ left it, not the start location of this layer. It then moves to to the start location, a similar start/finish location as the previous layer. 

If combing is disabled you can see the moves more clearly. 

Ive not ispected the gcode to see if there is in fact a small extrusion, but i dont see any reason for thre to be one. 

Looking at this I've also noticed that the wall ordering isnt being honoured. The outers are printing first. 

I noticed this because im printing this in TPU and its failed at a fimilar place twice, tanggled filament, I was looking at what might have caused it.

MariMakes, 2023-04-04:
Hey @orbitalr,

Thanks for your report 👍
I took a look but I'm at a loss. 😅 
I have no idea why Cura decides to comb like this, it makes no sense to me. 🙈 

I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞

Vandresc, 2023-04-28:
Devs see CURA-10551 for our internal reference