Ultimaker/Cura #14950: Crashing when opening project

sgtwebbpd, 2023-03-18:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Elegoo Neptune 3 Plus

### Reproduction steps

I imported an STL file and worked on the model, then saved the project as a 3MF file. I came back to open the project's 3MF, and Cura froze. I had to force close the program.

### Actual results

 I attempted this three more times with the same 3MF file and Cura froze each time. Then I deleted the file. Opened the STL in Cura, saved the project again. Closed Cura, and when I came back and tried to open the file again, Cura froze. 

### Expected results

When I try to open the 3MF file in cura the saved work on the project should load without freezing. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11009702/cura.log)
[Noble 6 Peter Scaled.zip](https://github.com/Ultimaker/Cura/files/11009704/Noble.6.Peter.Scaled.zip)

MariMakes, 2023-03-22:
Hey @sgtwebbpd,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I can successfully load and slice your Cura project in Cura 5.3.0
[image]

It might be related to a plug-in or the different printers in your configuration.
Would you mind sharing your configuration? 

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

MariMakes, 2023-04-03:
Hey @sgtwebbpd,

Thanks for sharing your config folder. 
I loaded it so to try to reproduce your issue, but I still can't reproduce your issue. 
I can load and slice your project successfully. 

I'm sorry but if we cannot reproduce your issue, we also cannot resolve it 😞 
I'll keep it open so we can keep track of other reports coming in. 

As a last resort you can try to on our nightly build. You can get it here: https://github.com/Ultimaker/Cura/discussions/15098