Ultimaker/Cura #14956: slicing dual color octopussy - error

gerardvandenelshout, 2023-03-19:
### Application Version

5.3.0

### Platform

windows 11

### Printer

Snapmaker A350

### Reproduction steps

slicing failed...and told me to report.. program stayed "life"..

### Actual results

stopped slicing with error

### Expected results

produced g-code

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[S2ADE_octopus_part1_origin met torens.zip](https://github.com/Ultimaker/Cura/files/11010602/S2ADE_octopus_part1_origin.met.torens.zip)

gerardvandenelshout, 2023-03-22:
> Hey @gerardvandenelshout <https://github.com/gerardvandenelshout> ,
>
> Thanks for your report 👍
> That looks like a super complicated model 🐙
>
> I'm getting an error loading your projectfile, my Cura doesn't recognize your printer.
> Did you make some alterations? Do you mind sharing your configuration folder?
>
> You can share your configuration folder by:
> 1. Go to Help at the top bar of Cura
> 2. Select Show Configuration Folder
> 3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
> You'll need to open Github in your browser to attach any files to your ticket
> You can click the view it on GitHub link at the bottom of your email to open GitHub in the browser.

Okidoki…doing than right now…. The printer is SNAPMAKER A350 DUAL HEAD (beta) by the way.

 

I will also include the octopussy

 

Kind regards and keep up the good work

 

Met vriendelijke groet | Kind regards,

Gerard van den Elshout
Backershagenlaan 2
2243AC Wassenaar / Netherlands

M: +31 6 5119 3204
E: ***@***.*** ***@***.***>

gerardvandenelshout, 2023-03-22:
Model is thingvers : https://www.thingiverse.com/thing:67503

As requested : 
[5.3.zip](https://github.com/Ultimaker/Cura/files/11040670/5.3.zip)
[5.3 rename in z01.zip](https://github.com/Ultimaker/Cura/files/11040694/5.3.rename.in.z01.zip)
[S2ADE_octopus_part1_origin with towers.zip](https://github.com/Ultimaker/Cura/files/11040741/S2ADE_octopus_part1_origin.with.towers.zip)

gerardvandenelshout, 2023-03-22:
Have provided the info you asked for… model is  https://www.thingiverse.com/thing:67503  and 

Printerprofile Snapmaker A350 dualhead (beta)

 

Curious to find if you take the model straight from thingivers with YOUR printersetup 
if you have the same issues ?

 

Kind regards

closed 2023-03-22 by gerardvandenelshout

gerardvandenelshout, 2023-03-28:
Thank you…will update.

 

Keep up the good work !

 

Met vriendelijke groet | Kind regards,

Gerard van den Elshout
Backershagenlaan 2
2243AC Wassenaar 

M: +31 6 5119 3204
E: ***@***.*** ***@***.***> 

 

I took another look at the files you shared.
I don't know why but the files seem to be corrupted for me.

Luckily we merged the Snapmaker dual extruder printers last week: #14624 <https://github.com/Ultimaker/Cura/pull/14624> , so I could use an internal version of Cura to test with.

I was able to slice the Thingiverse model with the default settings.
 <https://user-images.githubusercontent.com/40423138/228238596-dd3e4066-c495-4a8f-89e5-0b034a1a80ce.png> 

I also tried to copy over some of the setting you had in your project file and I was able to slice the model 🤔
 <https://user-images.githubusercontent.com/40423138/228237573-eb30869d-ba16-4d0b-b676-6ab93abc168f.png> 

Could it be related to one of the other settings you changed?
Are you able to slice other dual extrusion printjobs?
Perhaps like this one https://www.thingiverse.com/thing:2968129

highpowerxh, 2023-03-28:
@MariMakes if you rename 5.3 rename in z01.zip to 5.3.z01, and it should be able to extract along with 5.3.zip. That's the zip split achieve naming 
[image]
I'm not an expert on these definitions, #14624 is the first time I deal with cura machine definitions. but this extruder name seems not matching any existing machine definition and definitions in the configuration folder for me.
[image]
[image]