Ultimaker/Cura #14958: 5.3.0 Bad outer wall starts / Z seam

nm-8fw, 2023-03-19:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Anycubic Vyper

### Reproduction steps

So basically I sliced the same model with the same settings using Cura 5.3.0 and 5.2.2 and got bad Z seam with Cura 5.3.0.

### Actual results

The 5.3.0 one gave weird "smaller" Z-seam starts that look like little points in some places which resulted in gaps on the model.
[image]

[image]

### Expected results

Smooth Z seam as with 5.2.2 version. This is a screenshot of gcode made with Cura 5.2.2 and the prited model image.
[image]

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Calibration cubes with "Sharpest corner" Z seam setting printed from 5.3.0 and 5.2.2 gcodes.
[image]

orbitalr, 2023-03-20:
What order are the walls printing in your 2 examples, what's configured and what is it actually printing?

cetemadi, 2023-04-09:
I see the same issue on cylindrical prints even with _Seam Corner Preference_ set to None

cetemadi, 2023-04-09:
[CE5_Caps.3mf.zip](https://github.com/Ultimaker/Cura/files/11185002/CE5_Caps.3mf.zip)

superfloh247, 2023-04-11:
Ender 3 S1 plus + Sonic Pad also affected.

Cylindrical objects, so objects without corners - have little blobs all over the surface which should no be there.

[image]

00raq00, 2023-04-17:
For many years, Cura was considered the best slicer... It was...

MariMakes, 2023-05-04:
Hey @superfloh247,

Thanks for your report 👍
What you are seeing with the zits on a curved surface seems to be related to https://github.com/Ultimaker/Cura/issues/14811

We have a ticket on our backlog and our developers are working on the issue right now 💪 
It seems like Cura isn't filtering out microsegments correctly.