Ultimaker/Cura #14961: Slicing failed with unexpected error

ygoldfil, 2023-03-19:
### Application Version

5.3.0

### Platform

Linux 64bit Fedora 36

### Printer

Ender 3

### Reproduction steps

1. Trying to slice the attached STL file
2. Cura was invoked by GDK_BACKEND=x11 QT_QPA_PLATFORM=xcb LD_PRELOAD=/usr/lib64/libstdc++.so.6 /usr/bin/UltiMaker-Cura-5.3.0-linux-modern.AppImage

### Actual results

slice failed

### Expected results

expected to slice

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11011918/cura.log)
[perky-bad.zip](https://github.com/Ultimaker/Cura/files/11011920/perky-bad.zip)
[perky-good.zip](https://github.com/Ultimaker/Cura/files/11011921/perky-good.zip)

MariMakes, 2023-03-22:
Hey @ygoldfil,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

This mesh seems faulty
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/, or https://www.nanodlp.com/repair-3d-stl-file/.
Or you can use the "Make Solid" feature in AutoDesk MeshMixer. https://meshmixer.com/download.html
[image]

I was able to slice your model with the default Ender 3 settings.
[image]

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

ygoldfil, 2023-03-22:
Indeed the mesh was faulty as checked by mesh tools. Although earlier versions of Ultimaker did not crash. Maybe have the mesh tool mandatory and refuse to slice or warn ? Closing this bug in any case - Thanks !

closed 2023-03-22 by ygoldfil