Ultimaker/Cura #14963: Cura 5.3.0 - Maschine settings notice box at "Start G-code" & "End G-code" covers input & not scrollable

Gerhard-59, 2023-03-19:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Anycubic Kobra

### Reproduction steps

1.) When you hover with the mouse over the input fields of "Start G-code" & "End G-code" the notice box covers the whole window as long as you move the mouse outside. The notice box disappears when the mouse is outside this input fields.

2.) The notice box of  "Start G-code" & "End G-code" are not scrollable. They contain more information as what can be displayed in the box.
Please make them scrollable.

### Actual results

1.) There's no possibility to push the notice box away to have a reference for the possible input.

2.) They contain more information as what can be displayed in the box.

### Expected results

1.) Please make the notice box movable.

2.) Please make them scrollable.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

5axes, 2023-03-20:
It's a not a issue from Cura but from the plugin  **Cura Settings Guide** : 

https://github.com/Ghostkeeper/SettingsGuide/issues/89
https://github.com/Ghostkeeper/SettingsGuide/issues/108
https://github.com/Ghostkeeper/SettingsGuide/issues/105
https://github.com/Ghostkeeper/SettingsGuide/issues/103
https://github.com/Ghostkeeper/SettingsGuide/issues/95

To temporarily (or permanently) solve the problem, do not use  : **Show articles in setting Tooltips**  In the plugin Preferences

[image]

MariMakes, 2023-03-20:
Hey @Gerhard-59,

Welcome to the Ultimaker Cura Github 🚀 

This is related to the Settings Guide Plug-in.
People who don't have that plug-in installed, like me, just see this as tooltip:
[image]

You can temporarily disable your plug-in in the marketplace by clicking the gear in the top right corner and then disable.
If you are done editing your gcode you can enable it again. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

Gerhard-59, 2023-03-20:
Hello MariMakes,

Thank you for your fast reply.

As I don't want to modify this G-code before and after every day I can live with your suggestion to temporarily disable the plugin as a workaround.

But, in general the issue is not solved, because when someone wants to use this guide, it should work so that one can read and also copy the examples.

I would agree to shift this issue to the thread of the plugin.

sent from my Huawei P30 Pro with MailDroid

Gerhard Eckert
Oberweihersbucher Str. 5
90522 Oberasbach
Tel: +49 911 697886
mob: +49 173/8288 661
email: ***@***.***

https://github.com/Ultimaker/Cura/issues/14963#issuecomment-1476118032

MariMakes, 2023-03-20:
Thanks for getting back to us @Gerhard-59.
Good luck with editing the start/end gcode! 💪 
I'll close this issue.

closed 2023-03-20 by MariMakes