Ultimaker/Cura #14964: Flow is off by 10% and preview does not show it

dmitrym1, 2023-03-19:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Custom printer

### Reproduction steps

1. Slice a simple 20x20x20 cube with 0.4 mm line width. Single wall, some bottom, no top, no infill, brim

### Actual results

The printed cube wall width is 0.44 mm instead of 0.4 mm. Cura's preview shows 0.4 mm. PrusaSlicer's preview shows 0.44 mm.
Printed result confirmed on two different printers (highly customized FlyingBear Tornado2, MKS Gen L, klipper) belong to two different people

### Expected results

Printed cube wall, Cura's preview and PrusaSlicer's preview all have 0.4 mm wall

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Cura10%.zip](https://github.com/Ultimaker/Cura/files/11012291/Cura10.zip)
[image]
[image]

rburema, 2023-06-16:
While I don't fully trust either visualization (Prusa-preview shows wildly off widths for out 1st party printers for example), I think this does warrant further investigation.

I turned off the bottom and the adhesion as well, which left only the walls of the cube itself. By my calculations, this should have resulted in approximately 0.23282m infill used, assuming 1.75mm diameter. However, the resulting gcode-file shows 0.260759m of filament being used. (So about 11.5% more, which would fit with the +-10% you see in 'real life'.)

I'm not _completely_ willing to rule out other factors (including errors on my side), but I think we can make a ticket. (This doesn't mean it'll get picked up anytime soon probably, but it'll be in our system at least.)

(Devs: See internal ticket CURA-10696)

MariMakes, 2023-07-03:
Hey @dmitrym1,

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself. I'll be closing this issue, I hope you understand.

closed 2023-07-03 by MariMakes