Ultimaker/Cura #14970: cura 5.3 gcode error

qwerty8224, 2023-03-20:
### Application Version

5.3

### Platform

win11

### Printer

Gustom FFF

### Reproduction steps

as always cut the detail.

### Actual results

Bubbles on the model are caused by microscopic movements of the nozzle. It is observed on Ultimaker printers of different versions if you change the number of walls from 2 to 3

### Expected results

Fix ASAP.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[CFFFP_tronc_modifs_GITHAB.zip](https://github.com/Ultimaker/Cura/files/11017180/CFFFP_tronc_modifs_GITHAB.zip)

qwerty8224, 2023-03-20:
the problem goes away by increasing (maximum resolution) in Cura

orbitalr, 2023-03-20:
I had this start trying to remediate some issues I've had with some other prints. 

Custom printer setup (Klipper) running on a machine with buckets of resources. It looked like what happens when you push a board too far with too many gcode commands, moves start skipping and making sounds and appearance like this. 

Ive not tried again yet, I restarted my server hosting klipper and I'll try again. 

I had actually been slowing down the print speed. Image attached was faster then the one on the right. The one of the right has the issue. 

I'd not changed the resolution. I had been changing things like wall ordering, number, combing, but I honestly dont know what caused this. 

[image]

qwerty8224, 2023-03-20:
[орбитал](https://github.com/orbitalr) 
Thank you for showing the problem clearly. Play with the resolution and this problem is clearly visible in the slicer viewer. Do not rush there micro step or micro drop, good luck to you)))

orbitalr, 2023-03-20:
Ok, thanks, then our issue is not similar. In my example the slicer preview was normal, as expected.

qwerty8224, 2023-03-22:
Well, if it's okay, then close it.

discip, 2023-03-28:
@qwerty8224
@MariMakes pointed out exactly what is causing your problem!

discip, 2023-03-28:
@qwerty8224
@MariMakes pointed out exactly what is causing your problem!

So this can be closed. 😊

qwerty8224, 2023-03-29:
> @qwerty8224 @MariMakesименно то, что решение вашей проблемы!
> 
> Так что это можно закрыть.😊

Let it be so )))

closed 2023-03-29 by qwerty8224