Ultimaker/Cura #14976: Slicing failed

Dallan1985, 2023-03-20:
### Application Version

5.2.1

### Platform

MacOS Ventura 13.1

### Printer

Ender 3 pro

### Reproduction steps

Loaded attached file for slicing. Set the parameters. Clicked "slice". Error message appears. "Slicing failed".
[image]

### Actual results

Still fails

### Expected results

less failing, more printing.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Log and project files attached.
[cura.log](https://github.com/Ultimaker/Cura/files/11021725/cura.log)
[basalt_05 project.3mf.zip](https://github.com/Ultimaker/Cura/files/11021731/basalt_05.project.3mf.zip)

MariMakes, 2023-03-22:
Hey @Dallan1985,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

The new Cura 5.3 contains the fix for this for a number of slicing crashes. 

I was able to slice your model successfully in 5.3 Stable.
[image]

Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

If you still run into a crash, please double-check if you shared the correct file and let us know.
I'll take another look at what is going wrong. 👍 

Also I noticed that your mesh gave a faulty error. 
This mesh seems faulty
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/, or https://www.nanodlp.com/repair-3d-stl-file/.
Or you can use the "Make Solid" feature in AutoDesk MeshMixer. https://meshmixer.com/download.html

MariMakes, 2023-03-28:
Hey @Dallan1985,

That's so sweet, glad that you were able to safe the printjob. 
I'll close the ticket. Good luck with your print!

closed 2023-03-28 by MariMakes