Ultimaker/Cura #14978: Cannot select nozzle type

ansonl, 2023-03-20:
### Application Version

5.3

### Platform

Windows 10

### Printer

DXU UMO+

### Reproduction steps

1. I created a copy of the DXU UMO+ (dxu_umo.def.json) as (dxu2_umo.def.json) in the `AppData\Roaming\cura\5.3\definitions` folder with the changes being a new name and some minor machine setting overrides. Both definitions inherit from the DXU `dxu.def.json` definition.

2. When I launch Cura 5, I can select the original `dxu_umo.def.json` printer and select from available nozzle sizes, the DXU UMO+ has nozzle sizes available but the modified `dxu2_umo.def.json` printer no has nozzles available.

What am I missing here? Shouldn't the new modified definition import the `dxu_extruder` definition like its inherited `dxu` definition? 

original DXU UMO+ definition
[image]

modified definition
[image]

The modified definition file used was:
```
{
    "version": 2,
    "name": "UMO+ DXUv2",
    "inherits": "dxu",
    "overrides": {
        "machine_disallowed_areas": {
            "default_value": [
                [[100,  -102.5], [ 110,  -102.5], [ 110,  -62.5], [100,  -62.5]]
            ]
        },
        "machine_width": { "default_value": 220 },
        "machine_depth": { "default_value": 205 },
        "machine_height": { "default_value": 200 },
        "machine_head_with_fans_polygon": {
            "default_value": [
                [-46, 30],
                [-46, -34],
                [66, 30],
                [66, -34]
            ]
        },
        "machine_nozzle_heat_up_speed": {
            "default_value": 1.75
        },
        "machine_nozzle_cool_down_speed": { 
            "default_value": 0.8
        },
        "machine_nozzle_size": {
            "default_value": 0.4
        },
        "material_diameter": {
            "default_value": 1.75
        }
    }
}
```

### Actual results

Modified definition has no nozzle sizes available.

### Expected results

Modified definition should have same nozzle sizes available as DXU and DXU UMO+. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[stderr.log](https://github.com/Ultimaker/Cura/files/11022169/stderr.log)

ansonl, 2023-03-23:
@fieldOfView, Am I supposed to create a printer in Cura based on the existing definition and then manually edit the generated inst.cfg file? Wanted to confirm that's the intention. 

For variant files, there are already variant files for the original definition like (https://github.com/Ultimaker/Cura/blob/main/resources/variants/dxu/dxu_0.8.inst.cfg) which says `
definition = dxu`. 

Does a new printer definition that inherits from `dxu` not inherit the `dxu` variants and the inheritance only applies to the printer `def.json` files? 
So the correct way to add the new variants would be to copy the existing variants and and include the new printer definition filename in the variants' `definition`?

fieldOfView, 2023-03-27:
> Am I supposed to create a printer in Cura based on the existing definition and then manually edit the generated inst.cfg file?

Yes. The definition changes inst.cfg is normally where the settings you change end up in the "Machine Settings" dialog. But you can put just about any setting in there.

> Does a new printer definition that inherits from dxu not inherit the dxu variants and the inheritance only applies to the printer def.json files?

Correct.

> So the correct way to add the new variants would be to copy the existing variants and and include the new printer definition filename in the variants' definition

Judging from the huge number of variants that ship with Cura, that seems to be the idea.

MariMakes, 2023-04-03:
Hey all 👋 

Thanks for chipping in @fieldOfView. 
I am also a big fan of duplicating existing variants. Your configuration can already break by adding a space in the wrong place. 

Has your issue been resolved? 
Would you still like to keep this issue open or do you agree that we can close it?

ansonl, 2023-04-05:
@MariMakes and @fieldOfView. My issue has been resolved and this can be closed. Thanks for the help!

closed 2023-04-05 by ansonl