Ultimaker/Cura #14983: Refering to CLOSED Conical support missing #14789

Chad541121, 2023-03-21:
### Application Version

5.3.0

### Platform

Window 10

### Printer

Ender 5 Plus

### Reproduction steps

My ticket was closed without being resolved.
There is a setting that breaks the conical support brim.
No support brim is generated when the value is greater than zero.
I have figured out that when Generate Support is set to NONE, then support brim does not work.
[Speaker Cover no support brim when build plate adhesion set to NONE.zip](https://github.com/Ultimaker/Cura/files/11028155/Speaker.Cover.no.support.brim.when.build.plate.adhesion.set.to.NONE.zip)

### Actual results

No support brim is generated.
[image]

### Expected results

Model support brim set to none should not remove support brim.
I may be wrong, but I would think one should be able to have support brim without model brim. 
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This issue is in reference to a recently closed issue #14789 
Claiming the issue was fixed.  Please see above documentation showing the problem still exists.

MariMakes, 2023-04-03:
Hey @Chad541121,

Thanks for your report 👍
I apologize for closing your other bug,  I can clearly see what is wrong here. 
If buildplate adhesion is set to none, than there is no support brim. 

[image]

It seems that an easy workaround here is setting your adhesion to skirt.
Can you share why it's important that you adhesion is set to none? 

I'll bring it up to the team to see what they can do to improve it. 
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

MariMakes, 2023-04-06:
Hey @Chad541121,

I brought your issue up with the team. 👍 
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10502

Good Find! 👍

ZedOud, 2023-07-14:
I'm not sure if this was the case in 5.3.0 but in 5.4.0, Instead of "None" you can also trigger the bug by doing this:
1. Select "Skirt" for Build Plate Adhesion.
2. Set Skirt Line Count to 0
3. Set Skirt/Brim Minimum Length to 0

jxer, 2023-07-15:
wow this is a terrible response, why should i was filament using skirt if I don't need it? this worked in 5.1.1 without issue

MariMakes, 2023-08-09:
Quick update from our side.
We were able to resolve the issue here: https://github.com/Ultimaker/CuraEngine/pull/1917

That means that we can have a fix for this problem in the upcoming release.
Thanks for all the reports, apologies for the inconvenience 🙏

closed 2023-08-09 by MariMakes