Ultimaker/Cura #14984: Turning combing mode on messesup the time calculation

EmJay276, 2023-03-21:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Eryone ER20

### Reproduction steps

- Slice a part without combing
- Slice a part with combing

### Actual results

Slicing a part without combing give way more realistic results 
2h 25min for the z_drive_main_b_x2.stl (Voron 2.4) in my case.

Turning on combing reduces the time to 1h 53min
The real printing time is about 2h 30min

I have already seen this issue in 5.1

### Expected results

Calculate the time when using combing the same way than without combing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[EE_z_drive_main_b_x2.zip](https://github.com/Ultimaker/Cura/files/11028268/EE_z_drive_main_b_x2.zip)

MariMakes, 2023-04-03:
Hey @EmJay276,

Nice to meet you 👋 Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

That's an interesting issue 😮 
I guess we'll have to start some prints to see this for ourselves. 
How did you discover it? 

I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞

EmJay276, 2023-04-03:
Hey @MariMakes,

no problem I'm not in a hurry, the print takes as long as it takes ;)

I did some other tests/prints, with additional travel paths (several parts on one print plate) the time difference rises (up to factor >2 compared to disabled combing).
I think the travel paths (or the acceleration) is not taken into account with combing mode enabled.

MariMakes, 2023-04-06:
Hey @EmJay276,

I brought your issue up to the team. 👍 
There was a lightbulb moment with the developers who believe that one of our recent changes has not been included in the predicted printing time calculation (normally I would share the exact code here but I'm worried I would get it wrong) 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10501

Thanks for the report! 👍