Ultimaker/Cura #14987: Add printer wizard, tree controls open but don't close

pufferfish-tech, 2023-03-21:
### Application Version

5.3.0

### Platform

windows 10

### Printer

None

### Reproduction steps

1. Go to add printer. 
2. choose "non ultimaker printer"
3. Choose "non networked printer"
4. Try to fold up the "ultimaker" branch. 
5. It does not fold up. 
6. Expand another manufacturer eg elegoo
7. Try to then fold it back up
8. it does not fold up. 

### Actual results

Can't fold tree back up again no matter where you press

### Expected results

presumably pressing the arrow indicator should do that. It doesn't, nor does double clicking, or clicking in rage. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2023-04-03:
Hey @pufferfish-tech,

Thanks for the super clear report! Those make my day 😄 
Sorry, it took us a while to get back to you 😞 

I took a look and it seems that only one manufacturer can be opened at the same time. 
Opening one automatically closes the others. 
It seems to have been broken since at least since Cura 4.13 and you are the first report I see. 
Good find!

I'll bring it up to the team to see what they can do to improve it. 
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

fieldOfView, 2023-04-03:
> It seems to have been broken since at least since Cura 4.13

The list of printers has "always" behaved this way; the category that contains the currently selected printer definition is always open.

I agree that other places in Cura that use the opened/closed chevron will actually let you close the opened category, so it is not consistent behavior. But it is more or less a conscious decision ("by design").

casperlamboo, 2023-04-07:
Thank you for bringing this issue to our attention. This is indeed the intended behavior, and has been in Cura since as long as I can remember. That being said it is indeed a bit unexpected and dare I say annoying behavior. We decided and go ahead to and change the behavior, and should be fixed here: https://github.com/Ultimaker/Cura/commit/20b171dbebd6e19a990e1f0526cde7fa4724baea. The changes will be in our next release (version 5.4).

https://user-images.githubusercontent.com/6638028/230573550-df2f6607-4c11-4599-b1ee-51bf21a75724.mov

closed 2023-04-07 by casperlamboo