Ultimaker/Cura #14989: Cura does not recognize model is not at 0 Z

Scout8607, 2023-03-21:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

CR10S

### Reproduction steps

Set model at 5mm Z, sliced.
[image]
[CCR10S_Deathwatch_without_vizor.zip](https://github.com/Ultimaker/Cura/files/11034683/CCR10S_Deathwatch_without_vizor.zip)

### Actual results

Cura sliced model touching build plate instead of at 5mm Z.

### Expected results

Model should have sliced at 5mm with support material between 0-5mm.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Attached an image of preview as well.

Asterchades, 2023-03-22:
This looks to be changed behaviour between Cura 5.2 and Cura 5.3, and I'm not entirely sure it's deliberate.

Previously it would behave exactly as you are expecting. If you raise the model _and_ add supports, the model will be prepared in mid-air with a support structure underneath. To my knowledge this has been how it has functioned back as far as 4.2, if not earlier.

As of 5.3, this now appears to be obeying the `Remove Empty First Layers` setting, under `Mesh Fixes`. Trouble is that as soon as you tell it to generate supports or provide a model set to print as supports, this setting is no longer visible - even if it's being used. So the only way to toggle it is to remove the supports (a potentially frustrating task if you've created multiple pre-configured objects), change the setting, then restore the supports.

I'll flag this so hopefully a dev can take a look to verify whether the setting should be available (without the user editing their definitions manually) or whether the previous behaviour was the intended result.

nallath, 2023-04-11:
The team will take a look at this. Settings, when disabled, should no longer be used by the engine. In this case, that clearly happens.

MariMakes, 2023-05-09:
Hey All,

Quick update from our side.
This issue should be resolved with the next release.
The fix has been made here: https://github.com/Ultimaker/CuraEngine/commit/3b1f2e612534f0ed05e022159a5308a6d2c30479

Thanks for the reports ❤️
Together we make Cura better 🤝

closed 2023-05-09 by MariMakes