Ultimaker/Cura #14992: Another Slicing failed.

ThomasRahm, 2023-03-22:
### Application Version

5.3

### Platform

Windows 10

### Printer

Ender 5

### Reproduction steps

Try to slice.

### Actual results

Slicing fails, moving or rotating the model does not help. Enabling `Union Overlapping Volumes` enables the model to slice fine.
I am only reporting this one, because it is very consistently not working with `Union Overlapping Volumes` disabled.
[Hobgoblin.zip](https://github.com/Ultimaker/Cura/files/11035508/Hobgoblin.zip)

### Expected results

It slices.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
Also the system memory usage while it fails to slice is curious:
[image]

MariMakes, 2023-04-03:
OMG It's THOMAS! Hi 👋 
Sorry, it took us a while to get back to you 😓 

It looks like your mesh consists out of a number of submeshes. 
[image]

Have you tried doing fixing the normals with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/, or https://www.nanodlp.com/repair-3d-stl-file/.
Or you can use the "Make Solid" feature in AutoDesk MeshMixer. https://meshmixer.com/download.html

Can you let us know if this works?

ThomasRahm, 2023-04-07:
There is a reason i like to use this model for testing ;)

Using the Meshtools plugins and fixing normals did not fix it (same behavior).

The myminifactory fixer also did not fix it.
But there are differences in how it fails. 
The project file i appended crashes in https://github.com/Ultimaker/CuraEngine/blob/5.3/src/utils/VoronoiUtils.cpp#L199 after quite a while as `step_count` is for some reason much to large (`18213609930752`), so it just fills a vector until it runs out of memory.
With the myminifactory fixer it fails in https://github.com/Ultimaker/CuraEngine/blob/5.3/src/SkeletalTrapezoidation.cpp#L487

Meshmixer "make solid" causes it to slice, but it complains that the model is not watertight. 

I do not need the model to slice btw, i am reporting it as someone else may have another model that causes the same issues (and maybe this bug-report helps fixing some bugs)

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue: [CURAENGINE-1Y](https://ultimaker-o7.sentry.io/issues/4863562093/)