Ultimaker/Cura #14993: Will Not open on Main Monitor like all other apps.  Only will open on secondary Screen(Monitor)

lilrepete, 2023-03-22:
### Is your feature request related to a problem?

Will Not open on Main Monitor like all other apps.  Only will open on secondary Screen(Monitor)

### Describe the solution you'd like

Will Not open on Main Monitor like all other apps.  Only will open on secondary Screen(Monitor)

### Describe alternatives you've considered

Will Not open on Main Monitor like all other apps.  Only will open on secondary Screen(Monitor)

### Affected users and/or printers

EVERYONE

GregValiant, 2023-07-02:
Is this still a problem?
I will change it to a bug report.

lilrepete, 2023-07-03:
Yes, I just verified after all this time it still not changed…

 

Only app I have ever had do anything like this..

GregValiant, 2023-07-03:
Are both monitors the same brand/manufacturer?
What video card?

lilrepete, 2023-07-03:
2 Dell SE2717/HX Monitors

 

Intel® HD Graphics 530