Ultimaker/Cura #14995: Closing Outer Shell Gaps

CCPlusIT, 2023-03-22:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

CR6 SE

### Reproduction steps

After upgrading to new version from previous version and slicing a file previously used in older versions of Cura the gap between to faces has been removed and no longer prints the gap but makes it all one

[image]

[image]

Attached is the STL to try...

[Example-File.zip](https://github.com/Ultimaker/Cura/files/11038310/Example-File.zip)

It appears that something has changed since your previous release or some setting. Help would be appreciated.

### Actual results

[image]

### Expected results

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have attached a zip with the stl I have been using with previous versions with no issue.

CCPlusIT, 2023-03-23:
This seems to be just on the Creality printer profiles

MariMakes, 2023-04-03:
Hey @CCPlusIT,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I have the solution for you. 
You have to check the Print Thin Walls setting, it's not enabled by default. 
[image]

I know it's not intuitive, but it works for thin gaps as well.  😅 

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2023-04-03 by MariMakes