Ultimaker/Cura #14998: Slicing in 5.3.0 takes 7 minutes vs 4 seconds in 5.2.2

Snordraacher, 2023-03-22:
### Application Version

5.3.0

### Platform

MacOS Ventura 13.2.1 on a Core i9 2.3ghz Macbook Pro with AMD Radeon Pro 5500M & Intel 630

### Printer

Ender 3 Pro

### Reproduction steps

Slice the attached model with a skirt distance of 3mm, because it won't fit the buildplate otherwise

[Ender_3_PRO_PSU_fan_cover_honeycomb_wo_label.3mf.zip](https://github.com/Ultimaker/Cura/files/11040094/Ender_3_PRO_PSU_fan_cover_honeycomb_wo_label.3mf.zip)

### Actual results

Slicing seems to hang "forever", but after 10 minutes i checked the log and appearantly the process took 7 minutes to complete.

### Expected results

Slicing should be done in a matter of seconds, as it did before with 5.2.2
Always reproducible, but only with this project.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[logfile.log.zip](https://github.com/Ultimaker/Cura/files/11040216/logfile.log.zip)

MariMakes, 2023-04-03:
Hey @Snordraacher,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I'm having some problem opening your project file. My Cura says it's corrupted. 
[image]

I'm really interested in your report. 
Would you mind sharing it again? 

To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Snordraacher, 2023-04-19:
Hi, i don't have the project file anymore and i switched to an M2 based Mac so reproducing this is going to be difficult so it can be closed.

closed 2023-04-19 by Snordraacher