Ultimaker/Cura #15004: Releasing steppers in PauseAtHeight script leads to failed prints

Dim3nsioneer, 2023-03-23:
### Application Version

5.3.0

### Platform

Win10

### Printer

Ultimaker S5

### Reproduction steps

1. Load model
2. Add PauseAtHeight script
3. Slice
4. Send gcode to printer

### Actual results

- Printer prints the job until it reaches the pause
- Printer waits for one hour for the user to continue, then releases the stepper motors
- print bed of the S5 starts to lower due to gravity
- When the user resumes later than one hour after the pause started, the print bed can be lowered by several millimeters and printer then continues at wrong height resulting in a completely failed print

### Expected results

No disabling of the stepper motors so the print bed stays at the correct hight.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[devtest.zip](https://github.com/Ultimaker/Cura/files/11050901/devtest.zip)

gr5, 2023-03-23:
@Cycov take note.

Cycov, 2023-03-23:
Hi @Dim3nsioneer, auto-release after 1 hour was intended behaviour - see [line 469 in this commit](https://github.com/Ultimaker/Cura/pull/11718/commits/2ef483680263cd7d4a0f503d8b327e0758f4bc6a) 

It's a good idea to add another field where you can specify how much is the hold timeout up to a maximum of 32767 equivalent to about 9 hours, that should be enough. That number is picked because it's the maximum value of `int16_t` which is the minimum that the ANSI C++ guarantees an `int` to be and I couldn't find what is the maximum value that can be passed to the `M84` function - looked over in [the source](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/gcode/control/M17_M18_M84.cpp#L212).

I will make a new commit and pull request with this new change. Does that sound good @MariMakes ?

Dim3nsioneer, 2023-03-23:
It is beyond my undertanding why releasing the steppers during an intended pause is intended. I could only think of one reason: a 3d printer with stepper motors and/or drivers that overheat after some time. Such issues should be addressed in firmware imo and not in the slicer software.
Please enlight me if there is another reason.

Cycov, 2023-03-23:
@Dim3nsioneer, there is no reason besides what you just mentioned. At the time of implementation 1 hour was good enough as previously there was no timeout and the steppers disengaged instantly.
The point of the checkbox was an easy alternative to typing a big value in the "Disarm timeout" field and just give plenty enough time, so that when it beeps you can switch the filament without fear of moving the head. I didn't take into consideration other applications such as printing a part of a print and then leaving it to wait for more than one hour.

I do recognize that it's a bit misleading and by using that, one would expect that the printer stays engaged forever. However I do not know any GCODE that allows the motors to remain engaged forever so a timeout value must be specified.

As specified before, I could either:
- add a bigger timeout of 32767 seconds
- add a field so the user can specify the timeout so it's clear that it will disengage after a while
- change description/name to better reflect the behavior of the checkbox

gr5, 2023-03-23:
It should be "M84 S0"

I don't think this gcode (M84) was ever needed before - at least on ultimaker printers which use Marlin.  I could be wrong!  There are certainly gcodes to disable the steppers.  M17 and M18 enable/disable (respectively) the steppers.  M84 is for stepper timeout.  It's so that people can push the print head around by hand and to reduce annoying noise after doing a home or other movement unrelated to actual printing.  Typically people set it to values like 30 or 60 seconds.

M84 is typically not sent in a print job but used for example by octoprint when setting up the printer - when you first connect.  At least that is my guess.

M84 S0 disables the timeout.  It sets it to "infinite".  So that should be the gcode used.  It's okay to set it back to a reasonable value after the pause is complete (e.g. M84 S60) but I wouldn't mess with that.

Dim3nsioneer, 2023-03-23:
The fact that there is no gcode present to keep the steppers engaged does not mean that the steppers are disengaged. It depends on what the firmware of the printer is doing with the actual pause command, the M0.
And this is how it should be. The firmware of the printer should take care. Some printers might not need engaged steppers to keep the position. But it is a firmware thing and should not be overwritten by a slicer command.

AndreKR, 2024-01-24:
Do I see it correctly that there is no option to not generate `M84` at all? Even with "Keep motors engaged" it still generates `M84 S3600`. This is a problem with Klipper because it will turn the motors off immediately upon `M84` no matter what timeout is specified.

gr5, 2024-01-24:
> because it will turn the motors off immediately upon M84
Even if you do M84 S0?  Maybe this is a klipper bug.  What does Klipper do differently for an M18?

Sineos, 2024-01-24:
In fact, Klipper does not support additional arguments for M84 or M18.
Klipper supports the `[idle_timeout]` config option to customize actions upon and timeout for an idle event.

AndreKR, 2024-01-24:
I think the best course of action would be to add a method "Klipper" to the "Method" dropdown menu. If that is selected, most options should be hidden and `PAUSE` should be inserted as the pause gcode.