Ultimaker/Cura #15005: Slicing Failed

Spidrman54, 2023-03-23:
### Application Version

5.3.0 beta2

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

1. Loaded 

### Actual results

1. Loaded stl into Cura
2. Sliced
3. Received unable to slice error

### Expected results

The file should have sliced normally

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I did scale the stl down to 50% and it sliced fine. When I resized it to 100% and sliced it, the error occurred again. Please see the cura.logs for 3/23/23.

[image]
![Cura_slicer_afterError](https://user-images.githubusercontent.com/20644117/227219679-bb7d3665-7569-49f1-984f-3f32751502c5.pn
[cura.log](https://github.com/Ultimaker/Cura/files/11051340/cura.log)
g)

MariMakes, 2023-04-03:
Hey @Spidrman54,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

The new Cura 5.3 Stable contains the fix for this for a number of slicing crashes. 
Could you test if that's the case for your model as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Spidrman54, 2023-04-05:
I was able to slice the model perfectly with the stable 5.3.0 version.
Thank you!

Barry Beach

View my LinkedIn Profile at http://www.linkedin.com/in/barryvbeach

MariMakes, 2023-04-06:
Awesome so glad to hear! 😊 
I'll close this issue since it's resolved.

Happy printing

closed 2023-04-06 by MariMakes