Ultimaker/Cura #15006: Weird movements

masademan, 2023-03-23:
### Application Version

5.2

### Platform

Windows 19

### Printer

Creality Ender 3

### Reproduction steps

1. I sliced a model.
2. I previewed the gcode.
3. I saw the print head in the preview make weird random movements, like going back and forth.
4. It wasn't extruding, or going to a different part of the print, just random.
5. Happened in real life during printing.

### Actual results

The print head moves randomly and for seamingly no reason.

### Expected results

Random movements in previewed gcode, and in real life with the 3D printer.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11052499/cura.log)
https://drive.google.com/file/d/10v6f3kLSzfgB0UY24AVFXHwqME0LnOVc/view?usp=sharing

MariMakes, 2023-04-03:
Hey @masademan,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I'm really interested in your report, but I'll need a little bit more information from you.

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

masademan, 2023-04-07:
[Lizard tesallation.zip](https://github.com/Ultimaker/Cura/files/11180911/Lizard.tesallation.zip)

masademan, 2023-04-09:
[Lizard tesallation.zip](https://github.com/Ultimaker/Cura/files/11185621/Lizard.tesallation.zip)

closed 2023-05-29 by github-actions[bot]