Ultimaker/Cura #15008: Arc overhangs

masademan, 2023-03-23:
### Is your feature request related to a problem?

My feature is not related to a problem, but maybe many people can benefit from it.

### Describe the solution you'd like

I found 2 videos talking about arc overhangs very well.

The first one explains the concept of the arc overhang idea [here](https://www.youtube.com/watch?v=B0yo-o47688).

In the second video, it was explained how the community around PrusaSlicer made a sort of addition to make it easier to make arc overhangs, which can be found [here](https://www.youtube.com/watch?v=TGa_KvKLDR8).

I found both videos interesting and I believe that this could be a very interesting addition.

### Describe alternatives you've considered

None

### Affected users and/or printers

Pretty much everyone can use this feature.

GregValiant, 2023-03-23:
This is a duplicate of #14036 .  If you look at that request I tried to get this to work but failed.  The Gcode was OK but my printer (Ender 3 Pro) wasn't capable of printing like is shown in the videos.  Another thing is that it has to be printed REALLY slow.

5axes, 2023-03-24:
Dupliacte : https://github.com/Ultimaker/Cura/issues/14135