Ultimaker/Cura #15016: when "Build Plate Adhesion Type" is "None" then "Support/Support Structure" with "Tree" will disable "Enable Suppport Brim"

JoGrob, 2023-03-24:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

n/a

### Reproduction steps

1. Set "Support/Support Structure" to "Tree" and set "Enable Suppport Brim" to checked.
2. Set "Build Plate Adhesion Type" to "None"

### Actual results

Tree Support is generates without inner Brim (walls only).

### Expected results

Tree Support should be generates with inner Brim.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Tree Support BRIM is ok.
[image]

Tree Support BRIM is missing.
[image]

anonymousaga, 2023-03-26:
I got a temporary fix. I enabled the skirt, then set the distance to 1000mm. This means the skirt is not actually printed, yet the support brim does print. I was wondering why those supports failed a day ago, now I figured it out!

JoGrob, 2023-03-28:
See also issue repoprt: Support tree structure missing 1st layer #15034
https://github.com/Ultimaker/Cura/issues/15034