Ultimaker/Cura #15017: Extruder 1 Disabled but Still Selected when Printing

Luke-Houseman, 2023-03-24:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Bibo 2 Touch 

### Reproduction steps

1. I established a working cura slicer profile on an older version of cura (4.13.1)

2. I manually transferred each setting from my old profile to create a new printing profile on the slicer settings in Cura 5.3

3. I made sure to enable extruder 2 and set it as the active extruder in my cura profile, as my extruder 1 on my machine is currently in disrepair.  On the Cura UI on my computer, extruder 1 has been disabled. 

4. I load my sliced G code onto a SDHC card, safely eject the card after moving the file to the appropriate folder

5. Load the sliced project file to my printer and observe the displayed tempuratures

6. Extruder 1 is erroneously preheating and not extruder 2

7. Double check all settings, ensure that Extruder 2 is enabled and selected, and that Extruder 1 is disabled.

Any help or recommendation on a workaround for this issue would be appreciated.  

### Actual results

Extruder 1 instead of Extruder 2 is preheated when the file is loaded, printer prints using extruder 1.  

### Expected results

Extruder 2 is used for the entirety of the print, extruder 1 is not selected and stays cold for the entirety of the print.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Enclosure Lid Project Debug.zip](https://github.com/Ultimaker/Cura/files/11059966/Enclosure.Lid.Project.Debug.zip)
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/11060024/cura.log)
[cura.zip](https://github.com/Ultimaker/Cura/files/11060032/cura.zip)
[Enclosure Lid.zip](https://github.com/Ultimaker/Cura/files/11060133/Enclosure.Lid.zip)

I am not sure why the correct extruder is not being selected when the Cura UI seems to suggest that everything is set up correctly.  Please let me know if a solution to this problem is found.

MariMakes, 2023-04-04:
Hey @Luke-Houseman,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I took a look at your project.
I would also assume that your first extruder would not prompted to heat up.

Except that it semes to be mentioned in the Start and End Gcode. 
[image]

Can you remove the M104 T0 165 line from your Start Gcode and see if you are still seeing the problem?
Please let us know if that's not the case, than I'll take another look 😉

Drayson80, 2023-04-06:
I find a similar situation when selecting/disabling extruders of my Dual UMO. The reason is that the "Extruder Enabled" setting is not affecting the start code itself - unfortunately. Just the slicing.
In your setup it seems that the heatup is defined in the start code itself and not created by cura during slicing. Check in your oóutput code if the M104 lines are double.
my machine code looks like this:
[image]
the first lines including the heatup is created by cura and looks like this:
[image]

@MariMakes, 
from my perspective, the enable extruder setting shall have an effect on the global start code, especially affecting the priming and zeroing the extruder. (in my second screenshot you see the similar situation, only T0 is heated, but the priming is for T0 and T1)

Idea: split/introduce a seperate "extruder initialization g-code" per extruder, placed at the beginning of the machine start code (in addition to the existing extruder start g-code/end g-code). If one extruder is disabled, the respective initialization is skipped.

Guess that a lot of dual machines which handle the priming via gcode might be affected.  any chance that there is a fix on this soon?

closed 2023-04-19 by github-actions[bot]