Ultimaker/Cura #15018: Failed in following the steps of https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

alanlt, 2023-03-24:
### Application Version

Cura5.3

### Platform

Win10

### Printer

S5

### Reproduction steps

I want to run cura from source in Win10, and follow the Guide of [Running-Cura-from-Source](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source)
I failed in implement the step of 3. Initialize the Virtual Python Development Environment, 
when I implement the command of "conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv"
I don't know how to solve it? Can you help me?

### Actual results

PS D:\curaDev\Cura> conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
Configuration (profile_host):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
curaengine:compiler.cppstd=20
umspatial*:compiler.cppstd=20
scripta:compiler.cppstd=20
[options]
cura:devtools=True
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True

Configuration (profile_build):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
[options]
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True

arcus/5.2.2: WARN: Using dependencies specified in conandata.yml for version: 5.2.0-beta.2 while recipe is build for version: 5.2.2
WARN: arcus/5.2.2: requirement standardprojectsettings/[>=0.1.0]@ultimaker/stable overridden by pyarcus/5.2.2 to standardprojectsettings/0.1.0@ultimaker/stable
WARN: cpython/3.10.4: requirement zlib/1.2.11 overridden by pyarcus/5.2.2 to zlib/1.2.12
WARN: spdlog/1.10.0: requirement fmt/8.1.1 overridden by curaengine/5.4.0-alpha+testing_41@ultimaker/testing to fmt/9.0.0
WARN: savitar/5.2.2: requirement standardprojectsettings/[>=0.1.0]@ultimaker/stable overridden by pysavitar/5.2.2 to standardprojectsettings/0.1.0@ultimaker/stable
WARN: nest2d/5.2.2: requirement standardprojectsettings/[>=0.1.0]@ultimaker/stable overridden by pynest2d/5.2.2 to standardprojectsettings/0.1.0@ultimaker/stable
[... 171 lines cut ...]
pyarcus/5.2.2: CMake command: cmake -G "Ninja" -DCMAKE_TOOLCHAIN_FILE="C:/.conan/7cba0c/1/build/generators/conan_toolchain.cmake" -DCMAKE_INSTALL_PREFIX="C:/.conan/8781db/1" -DCMAKE_POLICY_DEFAULT_CMP0091="NEW" -DCMAKE_BUILD_TYPE="Release" "C:\.conan\7cba0c\1\."

----Running------
> "C:\.conan\7cba0c\1\build\generators\conanbuild.bat" && cmake -G "Ninja" -DCMAKE_TOOLCHAIN_FILE="C:/.conan/7cba0c/1/build/generators/conan_toolchain.cmake" -DCMAKE_INSTALL_PREFIX="C:/.conan/8781db/1" -DCMAKE_POLICY_DEFAULT_CMP0091="NEW" -DCMAKE_BUILD_TYPE="Release" "C:\.conan\7cba0c\1\."
-----------------
系统找不到指定的路径。
pyarcus/5.2.2:
pyarcus/5.2.2: ERROR: Package '46f4cc8d5d301cf7f04473c21ba4cacf2020349f' build failed
pyarcus/5.2.2: WARN: Build folder C:\.conan\7cba0c\1\build\Release
ERROR: pyarcus/5.2.2: Error in build() method, line 104
        cmake.configure()
        ConanException: Error 1 while executing cmake -G "Ninja" -DCMAKE_TOOLCHAIN_FILE="C:/.conan/7cba0c/1/build/generators/conan_toolchain.cmake" -DCMAKE_INSTALL_PREFIX="C:/.conan/8781db/1" -DCMAKE_POLICY_DEFAULT_CMP0091="NEW" -DCMAKE_BUILD_TYPE="Release" "C:\.conan\7cba0c\1\."

### Expected results

implement sucessfully.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[conan_Log.txt](https://github.com/Ultimaker/Cura/files/11060669/conan_Log.txt)

MariMakes, 2023-04-04:
Hey @alanlt,

😖 Sorry to hear you are stuck.
Is that still the case? 

I'm only a fledgling member of the Build From Source Club 👑 
Before we get started, I have a few questions to narrow down the troubleshooting:

1. Can you make sure you have specifically Conan version 1.56.0?  
`pip install conan == 1.56.0 --force `
You can check the [Conan Documentation]( https://docs.conan.io/en/latest/installation.html ) for the latest version and installation instructions

2. Did you try clearing the Conan Cache? 
`conan remove "*" -f `

3. Can you make sure you have the latest config with this command?
`conan config install https://github.com/Ultimaker/conan-config.git`

4. Can you make sure you have a default profile?
`conan profile new default --detect --force`

5. With these steps you should have a clean slate and you can follow the steps on our wiki again. 
[You can find it here. ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) 

If you are stuck in any way, please reach out. We are happy to help out. 
The best way to do that is to provide us with your complete run-out log. 
Not just the part that fails, but all the lines generated after you've given your command. 
We need it see your profiles and packages. 

Can you help me understand. What do you want to achieve by creating a Build?
Are you intending to share it with others or are you using it for research purposes?
I would love to know more about your project, maybe we can help.

alanlt, 2023-04-06:
Thank you for your anwer, MariMakes.
Now my problem was solved.  I think what I did which may be related are as below:
1. I specifically installed “MSVC v140-VS2015 C++生成工具”.  生成工具 means Generation tool.
2. I closed my firewall.
Then it works. I don't know which one is to solve the problem exactly.
And I'm confused in your Step 1 of the Guide:
**Installation
1. Configure Conan
_pip install conan --upgrade_
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force**
Since we are forced to use conan 1.56.0. Why does the Guide include "pip install conan --upgrade"? It will make conan upgraded to the latest version. Is it wrong or for any other special purpose?
Thanks.

MariMakes, 2023-04-06:
Glad to hear your issue is resolved. 

The upgrade was true for a while until there was a bug that broke our workflow. 
I was under the impression that it already was corrected, but it's clearly not. 
I'll ask the team to take a look 👍 

Good luck.

closed 2023-04-06 by MariMakes

stephenlang84, 2023-10-30:
> Hey @alanlt,
>
> 😖 Sorry to hear you are stuck. Is that still the case?
>
> I'm only a fledgling member of the Build From Source Club 👑 Before we get started, I have a few questions to narrow down the [...]

@MariMakes I'm trying to build Cura and CuraEngine on the same machine. As noted in [cura configure conan](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source#1-configure-conan), it requires minimum version is 1.62 to build Cura. Is there any solution to work with both Cura and CuraEngine and conan 1.62 and 1.56 ? 
Thanks for reading.