Ultimaker/Cura #15026: Z seam Alignment

Kris29us, 2023-03-25:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Cr-6 se

### Reproduction steps

Select z seam Alignment 

### Actual results

Results are very deference compare to version 4.13.1

### Expected results

Should be the same o better 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

[image]

MariMakes, 2023-04-04:
Hey @Kris29us,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

We made some improvements in the Zseam that will only show up if you select None as Seam Corner Preference. 
Can you try to slice the model with this updated setting? 

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

cetemadi, 2023-04-09:
I see the same issue on cylindrical prints even with _Seam Corner Preference_ set to None.
[CE5_Caps.3mf.zip](https://github.com/Ultimaker/Cura/files/11184999/CE5_Caps.3mf.zip)

closed 2023-04-19 by github-actions[bot]