Ultimaker/Cura #15028: Irregular flat surface slicing

NickBottelare, 2023-03-25:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Creality CR-6 SE

### Reproduction steps

Open a STL file of an object with a flat upper surface
Slice at 0.2 mm layer height, 5 bottom and top layers, line pattern. 
[CCR6SE_20230323 Filament lamp Basering 4.zip](https://github.com/Ultimaker/Cura/files/11069013/CCR6SE_20230323.Filament.lamp.Basering.4.zip)

I have made a new model from scratch, same issue.

[image]

### Actual results

The top layers (5 in this case) are irregular, which is not expected for a flat surface.

### Expected results

Top layers should be a regular line pattern.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

none

MariMakes, 2023-04-05:
Hey @NickBottelare,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

What a super interesting report! Thanks for sharing. 
It had me and some colleagues scratching our heads for a while, especially because it sliced seemingly normal for other printers like the UltiMaker S7.

We figured out that your issue is resolved by updating your Skin Expand Distance to 0.95.
But that would only solve this case. 
[image]

I'll bring it up to the team to see what they can do to improve it, because this bug might be an indicator of something else going wrong, but I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

casperlamboo, 2023-04-07:
Hey @NickBottelare,

Thank you once again for bring this to our attention. The behavior is indeed quite unexpected and undesired.

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10504

Thanks for the report! 👍

cutmoney, 2023-05-03:
I've figured out the "fix" of increasing the skin expand distance, but that has other undesirable consequences. Specifically, if I increase the value enough to fill the gaps, it creates over extrusion at the walls and results in bad surface finish. This was the first thing I did to fix the issue, then printed about 50 samples of a project to try to resolve my bad surface quality around the walls only to finally figure out it was that setting that was causing the issue. So, unfortunately, unless the value only needs to be increased slightly, it really doesn't resolve the issue.

MariMakes, 2023-05-16:
Quick update on our side. 
We are doing a big clean-up and noticed that this issue has been reported before. 

During clean-up, I noticed that this is a duplicate of https://github.com/Ultimaker/Cura/issues/12425

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12425

closed 2023-05-16 by MariMakes