Ultimaker/Cura #15029: Fail to slice after changing project content

Ananiaskye, 2023-03-25:
### Application Version

5.3.0

### Platform

Windows 10 Home

### Printer

Geeetech Mizar_S

### Reproduction steps

Sliced a project - exported G code to print (snake body)- Success.
Saved as project. 
Added 2 more objects (snake top head and rattle)
Adjusted print speed from 50 to 45mms
Adjusted print temp from 220 to 217
Removed some temps I previously enforced
Saved project
Saved new print setting profile - with new name (FlexiPla+_02)
Remove original STL ( already printed snake body)
Slice - Fail
Slice -fail

And here we are.

### Actual results

The project refuses to save as Gcode. 
[image]

### Expected results

Result in Gcode file to save. 
Here is a copy of successful project before I made the changes.
https://www.3dprintlog.com/prints/157759  <- you should be able to see it

Project files before and after the error also attached in comments. 
Cura log included.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11069491/cura.log)
[GMLYngRattleBody_precrash.zip](https://github.com/Ultimaker/Cura/files/11069493/GMLYngRattleBody_precrash.zip)
[YngRattleBody_postSliceFail.zip](https://github.com/Ultimaker/Cura/files/11069495/YngRattleBody_postSliceFail.zip)

I hope this helps, thanks for your work!

MariMakes, 2023-04-05:
Hey @Ananiaskye,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

The new Cura 5.3 contains the fix for this for a number of slicing crashes. 
I was able to slice your model successfully in 5.3 Stable.
[image]

Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

If you still run into a crash, please double-check if you shared the correct file and let us know.
I'll take another look at what is going wrong. 👍

Ananiaskye, 2023-04-21:
I was able to slice and print the model as well. 
I have not experienced this type of slicing error since but I now save my work more often. Thank you!

Ananiaskye, 2023-04-21:
This is closed.

MariMakes, 2023-05-04:
Glad to hear your were able to slice and print.
I'll close the issue 👍

closed 2023-05-04 by MariMakes