Ultimaker/Cura #15032: 'Uniform Scaling' toggles on and off when selecting models with <CTRL> + <A>

Chrissels75, 2023-03-25:
### Application Version

5.2.2 & 5.3.0

### Platform

Windows 11 22H2

### Printer

E3D Toolchanger (Custom printer profile)

### Reproduction steps

1. Open one or more models.
2. Select one or al of the models (it doesn't matter if you click one or select all models at this stage.)
3. Click the 'Scale (S)' button.
4. Notice if 'Uniform Scaling' is on or off.
5. Deselect all models by clicking on an empty space.
6. Select all models using <CTRL> + <A>.
7. Notice 'Uniform Scaling' changed.
8. When repeating steps 5 t/m 7, notice 'Uniform Scaling' changes everytime. It doesn't do this when selecting a model by clicking it, by selecting all models via right clicking and select 'Select All Models' from the menu or if you use the 'Edit' menu to select all models.

### Actual results

'Uniform Scaling' toggles on and off when selecting models with <CTRL> + <A>.

### Expected results

'Uniform Scaling' only turns on or off when you click it.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11069907/cura.log)

BTW 'Show settings folder' seems to be renamed to 'Show Configuration Folder' nowadays.

fieldOfView, 2023-03-25:
This is more or less by design; the ctrl key can be used to toggle the uniform option.

Chrissels75, 2023-03-25:
Using the CTRL key toggles it on or off temporarily when holding the key, when releasing the key 'Uniform Scaling' reverts back to the previous setting, but only if the 'Scaling' menu is open when you hit the CTRL key.
<CTRL> + <A> should be 'Select All Models', see the 'Edit' menu, it states 'Select All Models      CTRL-A', not 'Select All Models       CTRL-A and this also toggles 'Uniform Scaling' if 'Scaling' was the last menu you opened'.

Also, if you have 1 model selected, and press <CTRL> + <A> to select all modles, 'Uniform Scaling' doesn't change value.
Sorry, you can't convince me this is by design. This is a bug.

MariMakes, 2023-04-05:
Hey @Chrissels75,

Thanks for your report 👍
I took a look and I believe I understand your bug. 

You expect the Uniform Scaling checkmark to be show up consistently checked every time you select your models with CtrlA. 
Now it changes every time you select the models. I made a screen recording here.
[image]

I'll bring it up to the team to see what they can do to improve it. 
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

casperlamboo, 2023-04-07:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.
To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer/close this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-04-07 by casperlamboo