Ultimaker/Cura #15034: Support tree structure missing 1st layer

sthect, 2023-03-25:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Artillery X1

### Reproduction steps

Show picture with cura v5.2.2, 1st layer of support is present
Show picture with cura v5.3.0, 1st layer of support is missing
[image]
[image]

### Actual results

1st layer of support (tree structure) missing

### Expected results

1st layer of support (tree structure) present

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

JoGrob, 2023-03-28:
See also issue repoprt: when "Build Plate Adhesion Type" is "None" then "Support/Support Structure" with "Tree" will disable "Enable Suppport Brim" #15016
https://github.com/Ultimaker/Cura/issues/15016

MariMakes, 2023-04-04:
Thanks for your report 👍

I'll mark this as a duplicate for https://github.com/Ultimaker/Cura/issues/14983