Ultimaker/Cura #15035: Cura not opening files

damowatts, 2023-03-25:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender 5 plus

### Reproduction steps

Open application
Tried to open a file from application
[image]

### Actual results

Application hangs and becomes unresponsive

### Expected results

I get list of files the application can open

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11070228/cura.log)

MariMakes, 2023-04-05:
Hey @damowatts,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I recognize your issue, but this is the first time I've seen it reported on a Windows operating system. 
We released a Nightly Version which we hope should resolve issues like this, but it's hard to check for ourselves.

Could you take a look and let us know if this solves your issue? 
You can download the internal version here:  https://github.com/Ultimaker/Cura/discussions/15098

damowatts, 2023-04-05:
Thanks for getting back to me this concern resolved itself after a few
days. Didnt do anything different, i guess after turning my pc off and on
enough times it started working again. I did notice some settings had
changed so maybe something got corupted or a bad software issue

MariMakes, 2023-04-05:
Hey @damowatts,

Thanks for getting back to us. 
Glad to hear that it got resolved 😊 
I'll close this issue. 

Happy printing!

closed 2023-04-05 by MariMakes