Ultimaker/Cura #15037: Monitor not updating

vagredajr99, 2023-03-25:
### Application Version

5.3.0

### Platform

macOS Ventura 13.2.1

### Printer

Ultimaker S5

### Reproduction steps

1. Sliced model
2. Sent to printer
3. Viewed Monitor page in Cura

### Actual results

The Monitor page shows the previous print. 

### Expected results

Monitor page should show current print.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

While Cura logged me out, as it does quite regularly (certainly not long enough for a single print to finish -- not sure why timeout is set so arbitrarily low unless its to encourage subscriptions), even logging back in didn't update Monitor. Everything is up-to-date on the "Manage in Browser" page, of course.

MariMakes, 2023-03-31:
Hey @vagredajr99,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Your report reminds me of something reported internally. 
How do you slice your models? Do you load an STL/3mf or a Gcode? 

Can you also share why you prefer the Cura monitor over the Manage in Browser? 
What are you looking for when you are on the monitor page?

vagredajr99, 2023-04-01:
Hi @MariMakes no problem on the delay, that was pretty fast IMO :) I'm bringing in STL models only at this point, always have. As for the preference, I work online, from home, so having 800 tabs open makes it difficult to manage when I could just go to the Cura app, view the monitor, and see what's going on without having to wade through my open tabs otherwise. That's sort of the point of having a dedicated application vs. a web application, task-switching for people who are neurodivergent is a non-trivial event and I would prefer to save my brain energy for useful tasks vs. things that in my 30+ years of computing have always worked a certain way. I get having an always-available-anywhere-in-the-world web view, but if I had to do this with, say, Adobe applications, I would go slightly mad. Just because we can do a thing doesn't make it the optimal path :)