Ultimaker/Cura #15038: what causes this speed shift? It creates a hump in the outer wall

jahnj0584, 2023-03-25:
### Application Version

5.3 beta 2 

### Platform

windows 10

### Printer

ender 3 

### Reproduction steps

not sure tbh, it only happens on the bottom layers on the print.

Minimum layer time can be 1s or 12s or 23s and it is all identical. 

### Actual results

there's a decrease in feed along one section that causes a noticable hump when it switches feed rate.

### Expected results

a constant smooth feedrate along the outer wall 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]

MariMakes, 2023-04-05:
Hey @jahnj0584,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

This issue reminds me a little of https://github.com/Ultimaker/Cura/issues/13465, where a support mesh influences print speed. 

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

jahnj0584, 2023-04-21:
[print issue.zip](https://github.com/Ultimaker/Cura/files/11299167/print.issue.zip)
Here it is. Happens weather or not supports are enabled.

jahnj0584, 2023-05-04:
I guess it's still an issue in that it thinks that perfectly vertical wall portions are overhangs and need to be slowed down, and not just the little bit that actually is hanging over by a certain degree?

MariMakes, 2023-05-05:
Hey @jahnj0584,

I discussed your issue with the team. I asked them why the overhangs have a U shape, consisting out of 3 line segments instead of an I shape, a single line segment because that's the part that is actually the overhang. 
[image]

They explained that there are three line segments that are considered overhang because they start and stop in an overhang area. They extend so long in the wall because the wall is so straight in this model. It should be less of a problem in more organic or curvy models. 

I hope this helps 😇