Ultimaker/Cura #15039: Getting Error while slicing an object

malkhateeb89, 2023-03-26:
### Application Version

5.3.0

### Platform

MacOS

### Printer

Elegoo Neptune Pro

### Reproduction steps

1. i receive the error while slicing, I did nothing

### Actual results

Receiving the error slicing failed -  report a bug

### Expected results

get the sliced object to save the gcode

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11070773/cura.log)
[EN3PRO_Cobra_Vase_tex_v2.3mf.zip](https://github.com/Ultimaker/Cura/files/11070780/EN3PRO_Cobra_Vase_tex_v2.3mf.zip)

MariMakes, 2023-04-05:
Hey @malkhateeb89,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I'm getting the same crash as you, so I'll have someone from the team take a look.

I also noticed that I was able to slice your model if I rotated it 90 degrees, that often solves crashes that are caused by rounding errors. I hope that helps you get unstuck. 👍 
[image]

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker