Ultimaker/Cura #15040: birer birer yazdır hatası

mmnlgn, 2023-03-26:
### Application Version

5.3.0

### Platform

windows 10

### Printer

crealty cr10 smart pro

### Reproduction steps

daha önceki sürümde birer birer yazdır seçeneğinde açık gri alan birbiri içine girdiğinde hata vermiyordu.
bu sürümde ise hata veriyor.
örneğin; 112x12x3,6 ölçülerindeki parçadan solda beş sağda 5 olmak üzere 10 adet baskı alabiliyordum.
şimdiki sürümde hata veriyor.

### Actual results

seçenek=birer birer yazdır
112x12x3,6 ölçülerindeki parçadan solda beş sağda 5 olmak üzere 10 adet baskı alabiliyordum.
şimdiki sürümde hata veriyor.

### Expected results

seçenek=birer birer yazdır
112x12x3,6 ölçülerindeki parçadan solda beş sağda 5 olmak üzere 10 adet baskı alabililmek.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

MariMakes, 2023-04-05:
Hey @mmnlgn,

Üzgünüm, Türkçe bilmiyorum ama sanırım sana yardımcı olabilirim.

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I haven't seen any reports like this. 
What are you using for arranging the models on the buildplate?
What is your adhesion type? 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

mmnlgn, 2023-04-05:
Hello.
The source of the problem is incomplete information.
I had a problem when I chose the option to print one by one.
I confirmed "pin" each part to fix the problem. In this way, I placed the
pieces by giving them coordinates and succeeded in slicing them one by one.

Saygılarımla.
Mümin OLGUN

MariMakes ***@***.***>, 5 Nis 2023 Çar, 15:30 tarihinde şunu
yazdı:

MariMakes, 2023-04-06:
Hey @mmnlgn,

Thanks for getting back to us. 

In that case, this seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14112
We have a ticket on our backlog with the intent to improve this behavior.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14112
Thanks for your report 👍

closed 2023-04-06 by MariMakes