Ultimaker/Cura #15042: extra Infill wall count isnt affected by affected by infill layer thickness and weird behavior at 0 regular walls

fast2611, 2023-03-26:
### Application Version

5.3.0 (and lower)

### Platform

Windows 10

### Printer

Ender 3 Max (doesnt affect)

### Reproduction steps

extra Infill wall count isnt affected by affected by infill layer thickness, which would allow printing an internal wall only every second layer, as a use example.
(Additonally, setting outer walls to 0 seems to cause weird behavior with unrequested walls that are classified as top/bottom surfaces in the line type view, but I'm not sure if that part is just a setting I dont understand)

### Actual results

Considering extra Infill wall is flagged as infill in the viewer, has infil in its name, and is under the infill category, I think It not being affected affected by infill layer thickness is likely an oversight

### Expected results

Infill layer thickness should affect extra Infill wall count, in the provided example file that'd mean it would only get printed every second layer

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[image]
[demo.zip](https://github.com/Ultimaker/Cura/files/11071639/demo.zip)

MariMakes, 2023-04-05:
Hey @fast2611,

Thanks for your report 👍

Do I understand correctly that you want to have a Infill Thickness of two layers to reduce your infill printing time, but you also have an Extra Infill Wall Count to increase your printing time again? 
And the bug you are reporting is that the wall is printed every layer instead of every other layer like the rest of the infill. 
[image]

Can you help me understand why this is an important bug? 
Do you have an example of an actual failed printjob, since it only happens in the infill? 
Can you share what you are trying to achieve by combining these settings?

MariMakes, 2023-04-06:
Hey @fast2611,

Thanks for adding your motivation. 
That really helps us understand how people use our software and determine priority.

I'll bring it up to the team to see what they can do to improve it. 
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

fast2611, 2023-04-06:
<3
Since a bunch of the 3d type infills already print outer "walls"  if the inside of their patterns intersects with a wall adhering to the every other layer rule I cant imagine that theres much code that would need to be added, thanks a bunch for looking into it <3