Ultimaker/Cura #15043: [5.3.0] Retraction speeds / length not correctly applied for IDEX (and possibly other multi-extruder printers)

TheThomasD, 2023-03-26:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Tronxy Gemini S (based on Tronxy XY-2)

### Reproduction steps

See https://github.com/Ultimaker/CuraEngine/issues/1866 as I think this is an issue of the CuraEngine...

### Actual results

Retractions settings are not correctly taken into account.

### Expected results

Retractions settings are used as specified.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

A gcode file including the settings is attached in https://github.com/Ultimaker/CuraEngine/issues/1866

MariMakes, 2023-04-05:
Hey @TheThomasD,

Welcome to the UltiMaker Cura Github, we watch this repository more closely than the engine one🚀 
Thanks for your report 👍

This is outside my field of expertise, I'll have someone from the team take a look. 
I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞

TheThomasD, 2023-09-25:
Any update on the status?

MariMakes, 2023-10-19:
Hey @TheThomasD,

Sorry, it took us a while to get back to you 😞 
I'll poke the team again 😉