Ultimaker/Cura #15044: Erronious perimeter layers in Spriralize mode

jacquesglerner, 2023-03-26:
### Application Version

5.2 and up

### Platform

Windows 11

### Printer

Creality Ender 3 S1

### Reproduction steps

Have a "Vase" and tried to use  Special Modes and Spiralise Outer Contour. 
Every time I get some layers that pass through the volume instead of the expected perimeter layers.
[image]

### Actual results

It's not wise to print like that. if you use Creality 4.8.2 it doesn't happen.
[image]

### Expected results

Use the old Cura version of Creality flavor (4,8.2
[Brown_Paper_Bag_fixed.zip](https://github.com/Ultimaker/Cura/files/11072054/Brown_Paper_Bag_fixed.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I could upload more than one file, same bug.
[Brown_Paper_Bag_fixed.zip](https://github.com/Ultimaker/Cura/files/11072057/Brown_Paper_Bag_fixed.zip)
[watering_can_hollow_1.0_untested (3).zip](https://github.com/Ultimaker/Cura/files/11072068/watering_can_hollow_1.0_untested.3.zip)

GregValiant, 2023-03-26:
Thanks for the report and the files.
Spiralize was changed from the 4.x versions and is fussier about the model.  Your model is hollow and although it doesn't cause slicing to halt, it makes for extrusion paths across the center of a print as the tool path tries to move between separate islands (in this case the inner facing "Outer Wall" and the outer facing "Outer Wall".

The model needs to be solid through-and-through.  Since that is a new thing, "Remove all Holes" should be linked to spiralize and should turn on with spiralize.  If a user has un-linked them then they would need to be re-linked by clicking on the tool button next to the setting.  Take a look and see if Remove all Holes is on.  If not then turn it on and slice again.
One of the problems with the bag model is the saw blade" top edge.  That can't be spiralized because it devolves into a lot of disconnected islands.  There is no continuity from island to island and because there are gaps, Remove all Holes quits working.  You would need to put a support blocker in as a cutting mesh and keep the top edge from printing.  What I see with the paper bag appears to be "Expected Behavior".

The water can has the same "not solid through-and-through" problem.  In addition, Cura wants to put a hole in the side.  The hole causes the model to again devolve into 2 islands that are disconnected and so both Spiralize and Remove All Holes quit working at that point and don't start working again until the side hole closes.  So the 5.x versions of Cura will not spiralize that model.  That might be a setting issue but I wasn't able to get rid of the hole in the handle.  I'd call that a bug.  Someone from the Cura team will take a look.

Here is the water can sliced in 4.13.1
[image]

and here it is sliced in 5.3.0.  It doesn't even look like the same model.  
[image]

jacquesglerner, 2023-03-26:
That is my guess. And thanks for the explanation about the brown bag. I've done several water cans, even in fuzzy skin, which sort very well wonderful shapes but in "old style" versions, not anymore.

[https://www.thingiverse.com/make:917024](url)