Ultimaker/Cura #15046: Z Height issue with printer settings

Roch22, 2023-03-26:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Anycubic Chiron

### Reproduction steps

My Chiron has a Z height of 450mm but when I add a project of a height higher then 148mm I'm unable to slice.
Step 1- Add a project more than 148mm in Height
Step 2- Try to slice 

### Actual results

Error message " Unable to slice " 

### Expected results

Should be able to slice.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
<img width="560" alt="Screenshot 2023-03-26 095833" src="https://user-image
[image]
s.githubusercontent.com/128996874/227795027-2ef3227c-5fcd-41cf-9f64-4ca6834814cf.png">

Roch22, 2023-03-26:
[image]

Roch22, 2023-03-26:
[image]

Roch22, 2023-03-26:
I installed Cura 4.2.1 and with the same project I have no issues , it sliced it with no problem, For the version 5.3.0 I believe that all larger printers will have this height issue because I added other large printers and they have the same issues with the height, this is a serious bug.

Asterchades, 2023-03-27:
Make sure that you don't have the `Print Sequence` (`Special Modes`) set to `One at a time`. This will derive its vertical from the `Printhead Settings` rather than using the value given in `Printer Settings` to avoid having the gantry collide with models that have already completed. This limit is still imposed even if you only have a single model on the bed.

Alternatively if you set the `Gantry Height` to match your print volume height you can have it allow the whole volume. But this could easily represent problems should you try to print a multi-model print in future without addressing the setting.

Roch22, 2023-03-27:
Thanks for trying but it's still not working with the Gantry to match volume height and to set " One at a Time" I also only have 1 printer setup.
[image]

Roch22, 2023-03-27:
As you can see below the Z is at 199 which is stopping me to go higher
[image]

Asterchades, 2023-03-27:
Does it still do this when _not_ set to `One at a time`, as I suggested?

Asterchades, 2023-03-27:
Okay, something is wrong with how you have the newer one configured. I'm just not entirely sure what, as you've not provided a project file to determine exactly what settings you're using.

I will concede, however, that it's not the print sequence setting. I've added a Chiron to my local install, scaled a model to 450mm tall (also 290mm on X and Y), and it works exactly as you'd expect it should - regardless of the print sequence. So there's another setting that's been changed in your particular setup. As such a project file is going to be necessary to identify which one and whether it should be having this effect or not.

Roch22, 2023-03-27:
Ok thanks , do you have steps on how to do a project file?

Roch22, 2023-03-27:
[cura-bug.zip](https://github.com/Ultimaker/Cura/files/11080286/cura-bug.zip)

Asterchades, 2023-04-03:
Well, I have good news and I have bad news. Ironically they're both exactly the same thing: I'm able to slice the model, having loaded your project (thus printer configuration), perfectly. There's no height restriction at all whether it's printing sequentially or concurrently, under both the 5.3.0 alpha (which you appear to be using) and the 5.3.0 release build.

I'll flag this as "Under Investigation" so that hopefully someone more well versed in the inner workings of Cura can take a look. Unfortunately the issue is beyond my ability to diagnose.

Apologies for the delayed response.

Roch22, 2023-04-12:
Thank you for trying. 😉 Sent from Mail for Windows From: AsterchadesSent: April 3, 2023 4:28 AMTo: Ultimaker/CuraCc: Roch22; AuthorSubject: Re: [Ultimaker/Cura] Cura 5.3.0 - Z Height issue with printer settings (Issue #15046) Well, I have good news and I have bad news. Ironically they're both exactly the same thing: I'm able to slice the model, having loaded your project (thus printer configuration), perfectly. There's no height restriction at all whether it's printing sequentially or concurrently, under both the 5.3.0 alpha (which you appear to be using) and the 5.3.0 release build.I'll flag this as "Under Investigation" so that hopefully someone more well versed in the inner workings of Cura can take a look. Unfortunately the issue is beyond my ability to diagnose.Apologies for the delayed response.

Roch22, 2023-05-27:
Thanks again for all your help.
Did you hear of any success of fixing that issue because I’m still limited on what I can print in the layer height?
Is there a new version coming soon with a fix ? 
 
Thanks
Roch