Ultimaker/Cura #15051: Multiple 'seams' on curved surfaces

nirurin, 2023-03-27:
### Application Version

5.3

### Platform

windows 11

### Printer

Ender 5+

### Reproduction steps

Printed a ring cup-holder. 
Saw that it had... way more seams than it should on the finished print. Multiple seams per layer. 

The prrint file only has the usual single seam. While watching it print you can kinda see it stutter now and then, instead of doing smooth continuous circles. 

Does not happen in prusaslicer/superslicer (of course). I haven't yet tested on older versions of cura. 

### Actual results

Multiple 'seam' lines (or what look like seams) on each layer.
 
[image]
[image]
[image]

### Expected results

Should be smooth, with a single seam. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE5P_Ring_80mm.zip](https://github.com/Ultimaker/Cura/files/11073596/CE5P_Ring_80mm.zip)

discip, 2023-03-28:
Seems to be a duplicate of #14811!

MariMakes, 2023-04-05:
Hey @nirurin,

Thanks for your report 👍
I agree with @discip that this is probably a duplicate of https://github.com/Ultimaker/Cura/issues/14811
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14811