Ultimaker/Cura #15055: [5.3.0] Preview for "One at a time" print shows strange artifacts (gcode seems OK)

TheThomasD, 2023-03-27:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Artillery Sidewinder X1 (but similar issues with other printers)

### Reproduction steps

1. add multiple objects to build plate (I used two copies for each of the attached models)
2. switch to "one at a time" mode
3. switch to preview

### Actual results

As the screenshow shows, there are several strange lines shown originating from (0,0) after the third object (the axis at the back will be printed last). On other printers, a similar issue was obsevers, although sometimes with different "artifacts". Gcode seems OK. I haven't tested the print, but I didn't see any gcode commands with low or negative numbers, so I guess this is only a problem of the preview. I attached a project that shows the error already at layer 80.

[image]

### Expected results

Preview shows only correct 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[models.zip](https://github.com/Ultimaker/Cura/files/11076816/models.zip)
[project.zip](https://github.com/Ultimaker/Cura/files/11076848/project.zip)

MariMakes, 2023-04-05:
Hey @TheThomasD,

Thanks for your report 👍
I've seen this before in some very specific models on previous versions of Cura. 
But I'm having some trouble reproducing your issue:
[image]

I'm on Windows 11, and used your project file and steps. 
Can you still reproduce it yourself? Do you perhaps have different steps to reproduce?

TheThomasD, 2023-04-05:
Hi @MariMakes,

I just tested again with another printer (Anycubic Chiron). The result is the same (although the spacing of the objects is different).

Maybe you can try to just import the models I attached (both at the same time), select both and multiply them by 1, select "one at a time" and then slice? This worked for me.

[image]

MariMakes, 2023-04-06:
Hey @TheThomasD,

Thanks for the update. 👍 
I thought I recognized your issue, but I haven't seen it in a while 😅 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/11057
We have a ticket on our backlog with the intent to improve this behavior.
But don't worry, it doesn't seem to actually print the lines. 
Can you confirm if the lines are printed or not? 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/11057

closed 2023-04-06 by MariMakes

TheThomasD, 2023-04-06:
Hi @MariMakes,

thank you, yes, the other ticket you referenced describes the same issue (just for tthe Cura 4.x version). And from my findings, the gcode is also not showing the artifacts, hence, the strange lines are not actually printed.

Thanks!
Thomas