Ultimaker/Cura #15056: [5.3.0] Wall printing order used in Cura results in longer than necessary print times

TheThomasD, 2023-03-27:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Artillery Sidewinder X1 (but similar issues with other printers)

### Reproduction steps

1. take any model
2. slice (here: inner to outer wall ordering)
3. look at the wall printing order for inner, outer and extra walls

NB: this is independent of wether "outer to inner" or "inner to outer" is selected for "wall ordering". Also, "optimize wall printing order" does not have an effect.

### Actual results

1. Inner walls are drawn (requires moves to the different inner wall locations)
2. outer walls are drawn (requires moves to the different outer wall locations)

Sepending on how many walls you have, there will be a lot of travel involved.

Example:
[image]
[image]
[image]

### Expected results

In the older Cura versions, the wall ordering was different:

- print first wall inner walls
- print first wall outer wall
- print second wall inner walls
- print second wall outer wall
- ...

This required less travel between the walls as one wall was completely printed before another wall was started.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No attachment

TheThomasD, 2023-03-27:
To be fair, if there are several distinct "islands", it seems the wall order seems better (less traveling, see screenshot 1). However, expecially for first layers, less movement would make the first layer less error-prone (example of long walls in screenshot 2).

[image]

[image]

TheThomasD, 2023-03-28:
Just for reference: second layer in Cura 4.12 (with optimised wall printing order):
[image]

I'll attach a 5.3.0 screenshot of the second layer here as well (later).

MariMakes, 2023-04-05:
Hey @TheThomasD,

Thanks for your report 👍
This is probably a consequence of the Variable Linewidth Engine in Cura 5.x.
On the left you can see Cura 4.13 and on the right you can see Cura 5.3.

[image]

I don't really know how I can help you improve the wall ordering here. I'm not sure if it's even possible in Cura 5.x.
I'll bring it up to the team to see if they can help out. 
Fingers crossed 🤞

joncwchao, 2023-06-09:
Weighing in, this change made outer wall print quality much worse on the seam for flexibles (TPU in my case). In 4.13 printing each group of walls (2 inner and one outer) is one continuous line, not requiring a retract or extra move across walls. Now in 5.3, the 2 inners are extruded, then we move away and print other stuff, then come back, cross the inner walls, and start extruding the outer wall.
Even with relatively tuned settings the 5.3 seam is noticeably uglier with TPU.
I request a setting to restore the previous grouped inner-outer wall ordering.

Pla5ma, 2023-07-07:
I totally agree. We need at least a setting to group inner and outer walls together like in the old versions.

TheThomasD, 2023-07-07:
@MariMakes , any updates on this? For me it seems this is still an important issue. Do you have any info on whether this will be fixed at all, and if so, when I e could expect a fix?

Thanks! Thomas

Tripmaster7, 2023-07-11:
Wainting for the fix also - can you please give a timeline?

paselm, 2023-07-14:
I fully agree that outer and inner walls should be grouped together again!

Pla5ma, 2023-07-22:
Ultimaker should look into this issue asap because nearly every 3rd issue is related to inner and outer walls aren't printed together anymore. Dont you get it? Some examples: #16165, #15827, #15216. I can easily find many many more.

Arcari55, 2023-08-23:
I also experienced this problem.  I agree with the above posters that the default should be to group inner / outer walls together instead of printing the outer walls last.  I'm not sure what problem we are solving by printing the outer walls last.  If this feature is really something the community wants, maybe add the option in the configuration menu to print the outer wall last rather than have it hard coded.

I isolated the problem to a class in CuraEngine:  InsetOrderOptimizer.cpp   line 94:

[image]

Setting this to false returns the wall order back to the way it use to be calculated in 4.x

For anyone that wants to go off the reservation while waiting for the developers to decide what direction to go, I compiled a new version of the latest CuraEngine and posted it here:  [CuraEngine](https://github.com/Arcari55/CuraEngine-5.4.0.git)

This version will work with Cura 5.4.0 and is compiled for Windows 10  x64.  Download the .exe: build/Release/CuraEngine.exe and replace the version in your install (ProgramFiles/UltiMakerCura 5.4.0/) and restart Cura.

kastamauro, 2023-08-23:
> I also experienced this problem. I agree with the above posters that the default should be to group inner / outer walls together instead of printing the outer walls last. I'm not sure what problem [...]

Absolutely amazing, finally tree support working like a charm, z seam working as intended and wall order issue solved :D thanks for share.

Pla5ma, 2023-08-26:
> I also experienced this problem. I agree with the above posters that the default should be to group inner / outer walls together instead of printing the outer walls last. I'm not sure what problem [...]

I was so in hope but your version doesn't work for me. On many objects it will create completely messed up paths.

Arcari55, 2023-08-26:
> I was so in hope but your version doesn't work for me. On many objects it will create completely messed up paths.

Can you post an example and a project file (3mf) so we can see what's going on?

Pla5ma, 2023-08-26:
Here we go.

**Original Cura engine**

[image]

**Modified engine**

[image]

**Project files**

[3MF_Files.zip](https://github.com/Ultimaker/Cura/files/12445603/3MF_Files.zip)

Arcari55, 2023-08-26:
> Here we go.
> 
> **Original Cura engine**
> 
> [image]
> **Modified engine**
> 
> [image]
> **Project files**
> 
> [3MF_Files.zip](https://github.com/Ultimaker/Cura/files/12445603/3MF_Files.zip)
> 
> PS: I am realizing right now it happens on round corners only. Maybe arc welder isn't working properly with the modified version but I cannot test it because I am not at home right now.
> 
> PSS: But even if its arc welder it wouldn't explain all errors.

Ok its fixed, I removed an experimental piece of code that was modifying the extrusion path.  I posted the new version.  Thanks for the feedback.

Pla5ma, 2023-08-26:
Now it works like a charm!

This setting needs to be implemented into a Cura setting. Come on implementing a boolean as interface element is nothing.

MariMakes, 2023-08-28:
👀 I'm not a developer but I'll show the team. 💪 

@Arcari55 if you tag me in your pull request I'll make extra sure you get the credit for finding a solution.

casperlamboo, 2023-08-29:
Hi all, thanks for doing this investigation! I can give some insight why this was originally implemented like this. When printing we would like to limit the jumps in flow as much as possible. By frequently switching line types we differ in speed and thereby introducing more flow changes. The original strategy limits the number of flow-jumps by grouping toolpaths together that have similar flow characteristics.

That being said the example with the castle shows that frequent flow jumps do still occur in small island in the castle model. The flow jumps we are trying to prevent are negligible compared to these flow jumps. I went ahead and merged the proposed changes from @Arcari55 in this commit: https://github.com/Ultimaker/CuraEngine/commit/02c6ac4e5438d6d74f3b0c5d2c08b8e13c5ebc65. They will be in the next release.

closed 2023-08-29 by casperlamboo

TheThomasD, 2023-08-29:
Thank you all for your support in solving this issue! Looking forward to try this out in the next release :)

5axes, 2023-09-11:
> The flow jumps we are trying to prevent are negligible compared to these flow jumps. I went ahead and merged the proposed changes from @Arcari55 in this commit: [Ultimaker/CuraEngine@02c6ac4](https://github.com/Ultimaker/CuraEngine/commit/02c6ac4e5438d6d74f3b0c5d2c08b8e13c5ebc65). They will be in the next release.

We still have an issue with this modification, If you don't specify the Optimize Wall Printing Order then even if you have define the Wall Ordering from Inside to Outside you can have a completely random print order.

[image]

[Optimize.zip](https://github.com/Ultimaker/Cura/files/12578933/Optimize.zip)

pbruecken, 2023-09-20:
I'm not sure this is the same issue, but I sliced a 6 mm gear in Cura 4.12 from a Blender stl file and it printed.  The preview is below:
[image]

When I sliced it in Cura 5.4, it sliced like this:
[image]

When I printed the gears, the 4.12 gear printed successfully and the 5.4 gear was missing teath.  Both gears had to be reduced in size by 10% in Cura.

Pla5ma, 2023-09-20:
> We still have an issue with this modification, If you don't specify the Optimize Wall Printing Order then **even if you have define the Wall Ordering from Inside to Outside** you can have a [...]

This is not an issue and this is exactly for what optimize wall printing order was introduced. If it is disabled the printer order optimization behaves differently.

TheThomasD, 2023-11-07:
This does not seem to work even in Cura 5.5:

[image]

TheThomasD, 2023-11-07:
@casperlamboo ?

mottscanley, 2023-11-08:
> This does not seem to work even in Cura 5.5:
> 
> [image]
> 
> First, the inner walls are printed, then the outer wall. Am I missing something?

There is now another setting, called Group outer walls, under experimental settings. Turn that off.

https://github.com/Ultimaker/CuraEngine/blob/main/src/InsetOrderOptimizer.cpp#L96

TheThomasD, 2023-11-08:
Thank you @mottscanley , I wasn't aware of this one. Works as intended now!