Ultimaker/Cura #15058: [5.3.0] Unable to add a new printer

Ohmarinus, 2023-03-27:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Custom FFF

### Reproduction steps

Adding a new printer is filled with bugs.

Information in the dialogue is constantly reset, the software is unresponsive from time to time, etc. I cannot set my bed shape to elliptical, gcode flavor is constantly reset to 'Marlin', etc etc.

Steps:

Making a new printer:
[image]

After clicking 'add', the dialogue hangs for a couple of seconds.

Elliptic option has no effect as can be seen in the background:
[image]

Reprap firmware is selected now, but when I change any of the Printhead settings it will reset itself to Marlin and after changing a few values to zero, it will also reset the Printhead settings back to default:
[image]

[image]

There is also no way to expand the GCODE start and end input, so it's impossible to work with.

**Printhead settings cannot be changed, it constantly resets to the old values without clicking 'Next'.**

All in all I'm not even going to continue with this as it's no use.

**Here is a video that shows all:**
https://www.youtube.com/watch?v=C1PUgkalPlU

### Actual results

Settings are constantly reset to their default values and elliptic bed shape is not working. Gcode flavor resets to Marlin, printhead values keep popping back to their old values. All of this is happening without clicking 'Next'. It happens while I press 'tab' to go to the new input field, but it also happens when I mouseclick into the new input field.

### Expected results

I expect my values to stick in the input fields and actually work.

I'm not able to further work with Cura if it already goes wrong at the start. With Cura 5.2.1 I am not experiencing any of these issues.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Log files are not needed since slicing isn't the issue, nor is my project file.

Additional issues are that the mouse and camera movement isn't working properly, the mouse pointer moves a lot faster than the camera view is moved.

oliof, 2023-03-27:
I can confirm this bug on Linux (modern image). I also see this bug in 5.3b1. In 5.2.1 it works (with the old printer setup flow).

oliof, 2023-03-27:
The selected values are also changed on save when closing the machine settings.
The selected firmware flavor resets to Marlin when clicking the "origin at center" or "heated bed" checkboxes as well.

MariMakes, 2023-03-31:
Hey All,

Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14795
Our developers have taken a look and it should be resolved and released in an upcoming service pack. 
Apologies for the inconvenience. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14795
Thanks again for the report.

closed 2023-03-31 by MariMakes