Ultimaker/Cura #15061: Unable to slice STL file for unknown reason

TheDarwking, 2023-03-27:
### Application Version

5.2.2

### Platform

Windows 11

### Printer

Creality Ender 3 S1 Pro

### Reproduction steps

I am slicing a certain file, its happened on other files, which I always was forced to use other slicer for these files.  Cura works best for me so I would prefer to get these files sliced in Cura but its failing with an unknown error.  The other slicers slice no problem.  I have shared my project file of one such failed file.  

[CE3S1_Castle Base 15_32mm_cut2.zip](https://github.com/Ultimaker/Cura/files/11081830/CE3S1_Castle.Base.15_32mm_cut2.zip)

### Actual results

I click slice and I get an error about one second in.  No information is given to me on why.  I assume it does not like some shape of the file?

### Expected results

It should slice fine.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Please let me know if there is something missing or a way to fix this in a future release because I am tired of switching to Prusa which is having its own issues.

TheDarwking, 2023-03-27:
I did find for this stl, for some reason turning it 90 degrees on the build plate and it sliced fine.  But my printer is a square x/y build area and I don't know why in one orientation it would be better that the other when its the same first layer?  I am very confused so figured you could take a look.

MariMakes, 2023-04-05:
Hey @TheDarwking,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

The new Cura 5.3 contains the fix for this for a number of slicing crashes. 
I was able to slice your model successfully in 5.3 Stable.
[image]

Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

If you still run into a crash, please double-check if you shared the correct file and let us know.
I'll take another look at what is going wrong. 👍

TheDarwking, 2023-04-07:
I downloaded 5.3.0 and sliced the file and it worked.  Glad the new version fixed this issue.  If I get any more of these I will let you know.

MariMakes, 2023-05-05:
Hey @TheDarwking,

Thanks for the update 😊 
I'll close the issue. 

Happy printing!

closed 2023-05-05 by MariMakes